adafruit/pi_video_looper #127: Video + Image looper

monish6755, 2021-10-09:
I would like to play both Video and image in a looping. currently, it runs only video format in looping. could you help me to play image format as well in a single code. 
For example: in USD drive I am having both video and image files. when I plug it into raspberry pi it was playing only the video file. 
Could you make the required code change or assist me to make that possible. 
Thanks in Advance.

pherting, 2021-12-19:
Yeah i am looking for the same thing.. The problem is the overhead to create a video from images, thats not possible for a fast change of a single picture of a maybe daily base. But with omxplayer its not possible to play pictures, so you need a different software to show images and that means that you need a another logic to look is it picture or a video, than start this tool to show, wait and get the next media file. It's need a little bit more coding at this point ;)

tofuSCHNITZEL, 2021-12-19:
Hm I would argue the whole playlist, and file read logic is already there... One "just" needs to build a player class that can handle images... And we already use pygame for the background so maybe this would be not that difficult to do...

pherting, 2021-12-19:
Ah thanks, good hint! I am not family with python, more the nodejs guy but yeah this is a good weekend task for me ;)

tofuSCHNITZEL, 2021-12-19:
Yes ;) now I'm also thinking about it ;)

tofuSCHNITZEL, 2021-12-19:
I couldn't help myself and started with it... this is completely untested however:
https://github.com/tofuSCHNITZEL/pi_video_looper/tree/image-player
tomorrow I will set up a Rpi and debug it on real hardware

pherting, 2021-12-20:
Oh great!!! I will test it now.. i am trying at this moment other stuff also on my raspi, so it comes to a perfect timing! Great man, thank you very much!

tofuSCHNITZEL, 2021-12-21:
I just installed it on my pi looper and it works quite well... timing is not perfect but I already have an idea for some performance improvements...
there were some stupid bugs in my inital commit - so you should remove and reclone it now

tofuSCHNITZEL, 2021-12-21:
Boom, its done. v1.0.10 has an image player.

closed 2021-12-21 by tofuSCHNITZEL

pherting, 2021-12-21:
Thank you very much!
I try it, but i am to dumb.. there doesnt find pictures and reading from directory doesnt work also for me now.. ;)

tofuSCHNITZEL, 2021-12-21:
Did you do a fresh reinstall with the recommended image?

pherting, 2021-12-21:
1. No, i installed by hand.
2. Yes, i remove all folder, clone it new and run the install script =D

tofuSCHNITZEL, 2021-12-21:
What errors you you get in the supervisor logs

pherting, 2021-12-21:
`pi@raspberrypi:/home/pi_video_looper $ sudo tail -f /var/log/supervisor/*
==> /var/log/supervisor/supervisord.log <==
2021-12-21 20:55:02,378 INFO waiting for video_looper to die
2021-12-21 20:55:02,440 WARN received SIGTERM indicating exit request
2021-12-21 20:55:03,812 INFO stopped: video_looper (exit status 0)
2021-12-21 20:55:04,521 CRIT Supervisor is running as root.  Privileges were not dropped because no user is specified in the config file.  If you intend to run as root, you can set user=root in the config file to avoid this message.
2021-12-21 20:55:04,521 INFO Included extra file "/etc/supervisor/conf.d/video_looper.conf" during parsing
2021-12-21 20:55:04,556 INFO RPC interface 'supervisor' initialized
2021-12-21 20:55:04,557 CRIT Server 'unix_http_server' running without any HTTP authentication checking
2021-12-21 20:55:04,557 INFO supervisord started with pid 906
2021-12-21 20:55:05,565 INFO spawned: 'video_looper' with pid 909
2021-12-21 20:55:11,285 INFO success: video_looper entered RUNNING state, process has stayed up for > than 5 seconds (startsecs)

==> /var/log/supervisor/video_looper-stderr---supervisor-o3dTuq.log <==

==> /var/log/supervisor/video_looper-stdout---supervisor-oYagrl.log <==
pygame 1.9.4.post1
Hello from the pygame community. https://www.pygame.org/contribute.html
Starting Adafruit Video Looper.`

So nothing special...
I only changed:
file_reader=directory

This is working now after restart, but images from folders are ignored :/

tofuSCHNITZEL, 2021-12-21:
> So nothing special... I only changed: file_reader=directory

so you did not enable image_player at video_player setting in the ini?

pherting, 2021-12-21:
ohhh.... i doesn't see the new entry.. sorry :/
I removed the # on omxplayer and image_player.
Will it work together or only one at a time?

I test it with both, but that crashed with error: While reading from '/boot/video_looper.ini' [line 19]: option 'video_player' in section 'video_looper' already exists :/

The image_player alone works for me for pictures now.

My problem was also that a ./reload.sh is not enough for ini changes, ./install.sh is required after a change, correct !?

tofuSCHNITZEL, 2021-12-22:
> ohhh.... i doesn't see the new entry.. sorry :/ I removed the # on omxplayer and image_player. Will it work together or only one at a time?
>
> I test it with both, but that crashed with error: While [...]

only one kind of player can be active.

> My problem was also that a ./reload.sh is not enough for ini changes, ./install.sh is required after a change, correct !?

running reload.sh as root is enough. install is only necessary if you change the video_looper python code.

pherting, 2021-12-22:
Ok that's a pity, i was hoping i could create a playlist where videos and pictures can be processed. 
But thanks! Maybe i can create a mix of both ;)