adafruit/pi_video_looper #128: fatal error IL/OMX_Broadcom.h

hannile, 2021-11-11:
hi, 

i try to install the pi_video_looper on my raspberry pi 4 B and seems everything fine until i get to the point where i have to finish the installation process. gives me an error message: 

```
sudo ./install.sh
fatal error: IL/OMX_Broadcom.h: No such file or directory
fatal error: ilclient.h: No such file or directory
```

i am relatively new to this process and can't think of a solution, i tried to re-install the video_looper but failed.

tofuSCHNITZEL, 2021-11-11:
maybe your RPI install is broken.
reflash the SD (or use another SD) with a fresh raspian lite image. (using e.g. balenaEtcher)

hannile, 2021-11-11:
i reflashed the sd card with a fresh raspian desktop image, using the balenaEtcher; but still get the same error message

andrewcph, 2021-11-13:
Got the same problem. Tried flashing the same SD card with an older image I had from earlier and the pi_video_looper worked fine. But when flashing the same SD card with the new image from raspian (using RPI image v1.6.2) and I get the same problem.

tofuSCHNITZEL, 2021-11-13:
Maybe omxplayer does not work anymore for these newer images... Can you confirm the last version it was working with?

tofuSCHNITZEL, 2021-11-13:
Also omxplayer is abandoned...
So you need to use oder 32bit raspian versions.

omxplayer uses openvg for OSD and subtitles which isn't supported on Pi4. omxplayer uses openmax which has been deprecated for a long time and isn't supported with 64-bit kernels.

tofuSCHNITZEL, 2021-11-16:
cant reproduce the error you get... in the most recent rasperrypi os there is simply no package omxplayer anymore.
so the latest supported raspberry os image is 2021-05-28.

you can download it here;:
https://downloads.raspberrypi.org/raspios_lite_armhf/images/raspios_lite_armhf-2021-05-28/2021-05-07-raspios-buster-armhf-lite.zip

closed 2021-11-16 by tofuSCHNITZEL

arrowlab, 2022-01-26:
I faced the same issue but wanted as latest version as possible.
I reverted back to 2021 Oct 18 version (5.10.73):
sudo rpi-update 1597995e94e7ba3cd8866d249e6df1cf9a790e49

Then and re-run sudo ./install.sh
It can now install everything perfectly.

tofuSCHNITZEL, 2022-01-26:
Currently only the Legacy version of Raspberry Pi OS Lite is supported. You can download it from here: https://www.raspberrypi.com/software/operating-systems/#raspberry-pi-os-legacy