adafruit/pi_video_looper #129: Background Image doesn't reload after USBcopy

lordratner, 2021-12-01:
Hello hello!

First off, thanks for this neat program. I finally isolated an issue I was having. I have looper set up to copy the USB contents to the directory and also copy the background image loader.png. Upon inserting the USB drive, all videos and the loader.png file are copied over correctly. And the video list reloads to reflect the new set of videos in the folder and plays automatically. However there are two problems with the background image.

First, no matter where you have your background image (the default .ini uses /home/pi/static/bg.png), the loader.png from the memory key is loaded to /home/pi/loader.png. So the .ini needs to point to /home/pi/loader.png if you want to use background images loaded from the USB key. It would be better if loader.png on the USB key was copied to the path/filename specified in the .ini for the background image.

Second, after you fix the above issue (setting the image path to /home/pi/loader.png) the program does not load the newly uploaded image after copying it from the USB stick. Whatever the previous image was (including a blank screen if there was nothing) will be the background image until you restart the program.

I'm not sure how to fix it, but I'll dig around in the code and see if I can figure it out, but I suspect someone who is familiar with the program can fix both issues quickly. 

Thanks again!

tofuSCHNITZEL, 2021-12-01:
Hi,
since the path from where the looper picks up the background image is customisable I thought there was no need to also change the target of the png copy in copymode. so cou can just set `bgimage = /home/pi/loader.png`. tbh I use this image for something else that displays this image on bootup instead of the linux boot text.

The second thing is quite interesting and I will have a look on how to reload the bg image if one was copied.

Thanks for the feedback

tofuSCHNITZEL, 2021-12-01:
Hey,
could you please clone and try this version:
git clone --single-branch --branch copyfeature_update https://github.com/tofuSCHNITZEL/pi_video_looper
it should refresh the background image after copying

lordratner, 2021-12-02:
Ran the install per the Adafruit instructions, but with that branch, Installation finished but it wont start. When I run `sudo supervisor start video_looper` I get `video_looper: ERROR (spawn error)`

tofuSCHNITZEL, 2021-12-02:
Ups, stupid mistake. Please pull it now and try again

lordratner, 2021-12-02:
It runs now, but the background image still isn't updating after the USB downloads. I confirmed that the file is being replaced with the version on the USB drive. Restarting video_looper corrects it like before.

tofuSCHNITZEL, 2021-12-02:
Okay then I guess I need to do some testing... Will take some time

lordratner, 2021-12-02:
Thanks for the help. If there's anything I can do, let me know.

tofuSCHNITZEL, 2021-12-11:
sorry for the delay. was actually an easy fix - should have taken more time to think about the code in the first place... anyway its fixed in version 1.0.9.

cheers

closed 2021-12-11 by tofuSCHNITZEL