adafruit/pi_video_looper #130: Use of motion sensor?

fugazzy, 2021-12-01:
I am using video-looper for quite a while now to display small videos on a TV as information monitor. Quite often no one is arround and the TV could be switched off. 
This is possible by using 
```
TV On
echo on 0 | cec-client -s -d 1
TV off
echo standby 0 | cec-client -s -d 1
```
I would now like to use a motion sensor on the raspi gpio to switch the TV on and after a while off again.

Can that be done? Or can that be included into video-looper?

tofuSCHNITZEL, 2021-12-01:
I would recommend you write a separate little py script that you run as a service that sends the cec commands and reads the motions sensor - the video could keep looping in the background - or you stop/start the looper via shell commands also from the py script. but currently this is outside the scope of the pi video looper.
cheers,

closed 2021-12-01 by tofuSCHNITZEL

fugazzy, 2022-01-05:
I post my question here again, because this may be of interest for others also:
I had some time to works on this again and I got it working fine if started from console. I am struggeling to get it done automatically. It starts always fine, but I cannot get it stopped if started through supervisor.
Maybe someone here can give me a hint on a good start/stop script?
`
#!/usr/bin/python

# Bewegungsmelder ansteuern
# neue Methode ohne Polling

import RPi.GPIO as GPIO
import time
import datetime
import os
import subprocess
import syslog

print "BEWEGUNGSMELDER"
print ""

#Board Mode: Angabe der Pin-Nummer
GPIO.setmode(GPIO.BOARD)

#GPIO Pin definieren fuer den Dateneingang vom Sensor
PIR_GPIO = 16 # PIN-Nr auf dem Rapsi
#  GPIO als "Input" festlegen
GPIO.setup(PIR_GPIO, GPIO.IN)

# Initialisierung
starttime = datetime.datetime.now()
endtime = starttime

def period(delta, pattern):
    d = {'d': delta.days}
    d['h'], rem = divmod(delta.seconds, 3600)
    d['m'], d['s'] = divmod(rem, 60)
    return pattern.format(**d)

def MOTION(PIR_GPIO):
    global starttime # als globale Variable definieren

    if GPIO.input(PIR_GPIO):  # GPIO-Port = 1
      starttime = datetime.datetime.now()
      print  "%s - Bewegung erkannt" % time.ctime()
      syslog.syslog(" %s: Bewegung erkannt!" % time.ctime() )
      subprocess.Popen('/home/pi/tvon.sh', shell=True)
      #subprocess.Popen(/home/pi/tvwecken.sh', shell=True)
    else: # sonst
      endtime =  datetime.datetime.now()
      diff =  endtime - starttime
      diff = period(diff, "{h} Stunden, {m} Minuten, {s} Sekunden")
      print ("%s - Bewegungszeitraum abgelaufen! Zeitdifferenz: %s" % (time.ctime(), diff ))
      syslog.syslog("%s - Bewegungszeitraum abgelaufen! Zeitdifferenz: %s" % (time.ctime(), diff ))
      subprocess.Popen('/home/pi/tvoff.sh', shell=True)
      #subprocess.Popen(/home/pi/tvschlafen.sh', shell=True)
      print "%s - Warten auf Bewegung" % time.ctime()

# TV ausschalten / Schlafen legen bei Neustart
#subprocess.Popen('/home/pi/tvoff.sh', shell=True)
#subprocess.Popen('/home/pi/tvschlafen.sh', shell=True)

# VideoLooper starten
subprocess.Popen('/home/pi/vlstart.sh', shell=True)

print "%s - Start der Bewegungserkennung und VideoLooper" % time.ctime()
syslog.syslog("Start der Bewegungserkennung und VideoLooper")

try:
     GPIO.add_event_detect(PIR_GPIO, GPIO.BOTH, callback=MOTION)
     while 1:
          time.sleep(30)

except KeyboardInterrupt:
        GPIO.cleanup()
	print "%s - Stop der Bewegungserkennung und VideoLooper" % time.ctime()
	syslog.syslog("Stop der Bewegungserkennung und VideoLooper")
	# VideoLooper stoppen
	subprocess.Popen('/home/pi/vlstop.sh', shell=True)
`
vlstart.sh and vlstop.sh just contain
`sudo supervisorctl start video_looper
or
sudo supervisorctl stop video_looper`
this my supervisor config:
`# Supervisord configuration to run video looper at boot and
# ensure it runs continuously.
[program:pirvl]
command= /home/pi/pir-videolooper.py
directory = /home/pi
autostart=true
autorestart=unexpected
startsecs=5
`