adafruit/pi_video_looper #136: How to Randomize / Shuffle Videos in OMXplayer?

testataaaa, 2021-12-30:
Hi there,
Love it! So simple!
I was wondering how to get OMXplayer to play a random video on the USB stick, rather than playing them in a row?
I just can't get it working, not sure how to do this.

Does anybody have suggestions for this?

Thanks!

christiansievers, 2021-12-30:
just change the videolooper.ini file. 
This file is stored in the /boot part of the Raspberry Pi's SD card which means you can edit it directly on your computer.  Shutdown your Raspberry Pi, pull out the micro SD card, and insert it into your computer.  After your computer displays the files on the card open the video_looper.ini file in a text editor.

Find the line that says  "is_random = false“ and change it to "is_random = true“ . That’s it.

testataaaa, 2022-01-02:
Too easy! Thanks so much!

closed 2022-01-02 by tofuSCHNITZEL