adafruit/pi_video_looper #141: Image looper: first image is "unskippable" when display duration is high

millerscore, 2022-02-05:
Thank you very much for developing this project. I'm making an image projector for granny and having a solution with hot-swappable usb drives is a dream. Found an problem though:

-> Platform/operating system:
Raspberry Pi 3 model B with DietPi v7.8.2, premade image by the man himself @christiansievers from https://videolooper.de/ - v.2.5 – Jan 2022 

-> Repro Steps:
- Take image, flash to sd card, set "duration" of [image_player] in video_looper.ini to a high value, e.g. 100 (seconds)
- Set up rasbpi with hdmi screen, keyboard and usb drive with at least two images on it
- Boot
- Upon seeing the first image on screen, press "k" on keyboard to skip to the next
- Observe unusually long black screen before the next image is displayed

Skipping images other than the first image (before looping) works fine.

tofuSCHNITZEL, 2022-02-06:
Thanks for reporting this. I will check it out.

christiansievers, 2022-02-06:
Hi, the image is just the packaging, you’ll get exactly the same result when you install the script yourself. The image_player is still in beta - maybe @tofuschnitzel can help?

tofuSCHNITZEL, 2022-02-06:
My guess is this is also happening with video files and the bug is  in the main playlist logic and therefore independed of the player type. Will get back to you.

tofuSCHNITZEL, 2022-02-07:
fixed in [v1.0.11](https://github.com/adafruit/pi_video_looper/releases/tag/v1.0.11)
thanks for reporting!

closed 2022-02-07 by tofuSCHNITZEL

millerscore, 2022-02-18:
You're welcome, thanks for fixing