adafruit/pi_video_looper #142: Video looper not working with hdmi screen

bassembrace, 2022-02-12:
Hi,

I have bought a LCD round screen, it boots into the desktop GUI mode perfectly but seems not to turn on while in command line mode / video looper mode. What could be the issue? I already tried forcing hotplugging, different resolutions, hdmi_cvt fixes, etc. I even followed this in depth article, but nothing helped: https://learn.adafruit.com/using-weird-displays-with-raspberry-pi/everything-else

I would be very grateful for solving this, as video looper works on a "normal" hdmi desktop screen, but the command line view / video looper section turns off the screens.

christiansievers, 2022-02-14:
Which screen is that - do you have any documentation? What kind of Raspberry do you use?

dannpye, 2023-01-20:
> The video_looper and the HyperPixel Round 2.1 works very well together, very straightforward installation process and never issues with video_looper, recommendation. Example here: [http://rootfriend.com/videoplayer-kompakt]

Hi shell3000, I noticed you got the round screen working, how did you do this? It doesn't work for me, I just get blank screen with video looper loads.

shell3000, 2023-01-23:
Legacy Lite OS plus the Pimoroni-Setup <https://shop.pimoroni.com/products/hyperpixel-round?variant=39381081882707>. Built some Hyperpixel-Screens before, never a problem.

closed 2023-11-16 by tofuSCHNITZEL