adafruit/pi_video_looper #143: Keyboard command to skip backward

millerscore, 2022-02-18:
As mentioned in a previous issue, my use case is to make an image projector for granny. I would use the image looper and set the duration to about 20 minutes, so her photos change every now and then or you could use the skip forward key to go through an album while sitting together and talking (where you probably never reach 20 minutes per image).

When you are sitting together watching photos, a button to skip backward is just as important as a button to skip forward. It's pretty much essential, somebody always has something to say about the photo you just skipped.

If possible, I would really like a keyboard command to do that, similar to "k" to skip forward. It is probably worth considering how this command should affect the video looper, either restarting the current one or skipping back to the previous video.

millerscore, 2022-04-11:
Is this feature realistic or do you consider it out of scope? I get that there is a good argument for keeping things simple in this project.

tofuSCHNITZEL, 2022-04-11:
its coming... ;)
I just whipped something up real quick:
https://github.com/adafruit/pi_video_looper/tree/feat_prev
beware! this is untested for now because I need to setup a testinstall first

millerscore, 2022-04-12:
Well I'll be damned, thank you :) I'm not in a place where I can test it on my raspbi atm, maybe in the next few weeks, but keep me posted please.

tofuSCHNITZEL, 2022-04-13:
my quick tests now seem promising... need to test it with all different combinations of players still.
but I would like your opinion on something:

skipping back clears the play count of all files because if you use the "repeat" function via the filename it would go back to the e.g. 2/3 play count of a file and this would not make sense right?

secondly:
currently the skip function skips only the current iteration of a file - so if a file is repeated 3 times pressing skip only skips one play count and not immediately to the next file -
what would you say makes more sense? to have the skip function skip to the next file or the next "video according to the playlist, which in case of a repeated file would be the next play count of that file)

millerscore, 2022-04-14:
Even though the repeat function is not in my use case, here are thoughts. Short answer: Both forward and backward skip should behave the same imho, either they both step through iterations or both "ignore" the repeat function and always switch to another file. I personally prefer file skip over repeat iteration skip, but I can imagine some users might disagree. 
Long, coffee driven answer below 😄

I think for a function called "skip" it makes the most sense to ignore repetitions and play "something different" next. It's what I would expect just based on the name of that function. Your mileage may vary. But, this even opens up a new use case where you play one video "indefinitely" (4095 times or sth) until you *want* to skip to the next. That, however, could lead right to the next feature request to use "-1" or something similar as repeat parameter value to do true endless runs 😅

Depending on how much you see the danger of feature creep, I personally wouldn't *mind* skipping through repetitions instead of files *at all*. It is not fundamental in my opinion, both solutions work and that's what matters in this project as far as I can tell.

Another thing to keep in mind here is, as a user I would appreciate it if skipping back and forward behave similarly. So either they both step through iterations or both ignore repetitions and always switch to another file. That makes sense and feels consistent. But at the end of the day I'm happy to have any option to control the basic playback in both directions. The amount of effort it takes to streamline the features is entirely up to you. I'm not a client here, just a humble requester. 🙃

tofuSCHNITZEL, 2022-04-14:
Thanks for the feedback! and I totally agree that the most important thing is that they behave the same.
at the time when I implemented the skip function it would have been more work to make it skip to the next file instead of iterations so its actually a bug (=feature) that it only skips the iteration.
on the other hand implementing that the "back" function only goes back through iterations (and when reached the 1st iteration jumps back to the previous file) would also be complicated.

so I will make skip behave like the back function with the release of the back feature.

just need some more testing to make sure I did not break anything ;)

millerscore, 2022-04-19:
I got back to my hardware and just did a fresh install from the feat_prev branch, looks good to me 👍 but I'm just testing for positives here.

tofuSCHNITZEL, 2022-04-19:
great news. thanks for checking it out!

millerscore, 2022-04-25:
I used image_player, yes

tofuSCHNITZEL, 2022-09-13:
now its official ;)
https://github.com/adafruit/pi_video_looper/releases/tag/v1.0.12

closed 2022-09-13 by tofuSCHNITZEL