adafruit/pi_video_looper #144: Issue with copy mode not copying all files from USB drive

Lunal-Tic, 2022-03-01:
Ran this issue past the folks over on the [Adafruit forum](https://forums.adafruit.com/viewtopic.php?f=8&t=188754&p=913700#p913545) but appear to have hit a wall so thought I'd post here.  Please let me know if is any additional information required.  I am new to this so it is likely I've omitted things.

I'm running a Rpi 3B+, Samsung EVO 64GB sdcard, and Video Looper v2.5 from https://videolooper.de/   The files are being copied from a 32GB USB drive with 233 video files (24.9GB).  The system is planned to display 12 hours a day however the Rpi will not be turned off only the monitor will be shutdown by timer.

The idea was to use copy mode to get videos to the SD card since I've got the room there. However it will only copy 6 videos of 200+ to the SD card and then goes into a boot/copy loop after showing the status bar partially completing the 6th video. If I stop the process and run the system without the USB drive it will loop the videos it copied.

Here is the way I have it set:
```
# Note that you still need to change the directory by modifying the setting in
# the [directory] section below.  The default is usb_drive.
file_reader = usb_drive
#file_reader = directory
#file_reader = usb_drive_copymode
```

and

```
# Directory file reader configuration follows.
[directory]

# The path to search for movies when using the directory file reader.
path = /home/pi/video
```

I am unsure if there are other settings that I missed or if I should change the setting to file_reader = directory once the USB files are copied.  Also I tried both with and without a password setting for the USB.

Additionally I've used different USB drives, different SD cards, and different numbers of video files to copy. When I only had 10 video files it would copy 3. The system will run fine if the videos are on the USB drive.  I've fresh installed video looper each time I tried a new variable which is why I'm pretty sure it's working OK when it runs from a USB drive with an unmodified video_looper.ini file. (only let it run for about 10 videos)

Two questions:
1. Is there some way to copy as many videos as is there is space on the SD card?
2. Should I even be using the SD card to play the videos? I've seen a [thread](https://github.com/adafruit/pi_video_looper/issues/89) that claims some issue doing that.

TIA
-LT

tofuSCHNITZEL, 2022-03-01:
Hi,
thanks for sharing this issue..
To be honest I have never tested the copymode with that many videos or this big of a total size...

I will try to create a similar test scenario with lots of files and some gigabytes of files...
can you please share the supervisor log output as described in the readme?

thanks

Lunal-Tic, 2022-03-01:
Thank you for the quick reply.  I've over written the SD card with a stock version of looper so I will have to recreate it from my end as well.  I'll take a look at the readme and post the supervisor log once I'm able to get it back to that state.

I'm in Tokyo so apologies for any delay in replying due to time zones.

Cheers,
-LT

Lunal-Tic, 2022-03-01:
I am unable to find information about the supervisor log in the 3 readme files on the SD. (videolooper_README.txt, dietpi-README.md, and overlays\README)  

The videolooper_README.txt has only this information:

```

__   _(_) __| | ___  ___ | | ___   ___  _ __   ___ _ __        __| | ___ 
\ \ / / |/ _` |/ _ \/ _ \| |/ _ \ / _ \| '_ \ / _ \ '__|      / _` |/ _ \
 \ V /| | (_| |  __/ (_) | | (_) | (_) | |_) |  __/ |     _  | (_| |  __/
  \_/ |_|\__,_|\___|\___/|_|\___/ \___/| .__/ \___|_|    (_)  \__,_|\___|
                                       |_|                               

Christian Sievers 2015-2022

www.videolooper.de

v2.5 (using pi_video_looper v1.0.10)
```

tofuSCHNITZEL, 2022-03-01:
if enabled (via config file) log output can be found in /var/log/supervisor/ You can use e.g. tail -f /var/log/supervisor/* to view the logs

tofuSCHNITZEL, 2022-03-01:
Follow the readme from this repository (https://github.com/adafruit/pi_video_looper) and install it on a regular legacy RPI OS Lite. 
or try to enable console output in the ini of your installation then SSH to the rpi and check the supervisor folder mentioned above.

foocp, 2022-03-01:
You could follow the instructions in the original Adafruit tutorial: [tutorial](https://learn.adafruit.com/raspberry-pi-video-looper?view=all) to install it manually (i.e. first install a fresh Raspberry OS, then the looper script). 

A quicker way could be to SSH into your running looper system. The videolooper.de site (which is this script here on Github but packaged) gives some instructions to do that. Not sure if you can enable the supervisor log on that system, though.

Lunal-Tic, 2022-03-02:
Thank you both for the info and suggestions.  I'll read up on things and get back as I am able.

Lunal-Tic, 2022-03-02:
It appears I might have proved that different installations are in fact different.  Following the manual install method in the link provided above, without the hello_video player, I was able to get the copy mode to function correctly.  It copied 231 video files (corrected from 233 above) from the USB drive to the SD and has been playing them back for the last 20 minutes.  For reference the video files vary in size from ~7MB - 800MB  I don't know if it would help but I could upload 10+ of the video files I'm using for testing purposes.  I'm afraid I don't have online storage large enough for the entire 231.

Just to verify I took a second 64GB SD card and reinstalled Video Looper v2.5 from https://videolooper.de/ Then made the changes to video_looper.ini as mentioned above.  The result was exactly the same as my initial issue.  6 files of 231 copied and then would kick out to reboot.  If I pulled the USB during this time, before the system started to copy files, it would show the countdown screen and how many videos were available before playing.  It appears that the issue is specifically with the Video Looper v2.5 installation (using pi_video_looper v1.0.10 as stated above).  Would it be possible to see if we could get this particular installation working properly?  

Additionally I was never able to find or figure out how to get the supervisor log.  It is mentioned at the bottom of the Installation section [here ](https://github.com/adafruit/pi_video_looper)but doesn't say how to get it activated.  I did find something about dtdebug [here ](https://www.raspberrypi.com/documentation/computers/configuration.html)but I'm not sure that's what you are referring to.

tofuSCHNITZEL, 2022-03-02:
Just something that came to my mind. Maybe your image @christiansievers does not expand the partition to the whole sd card and after copying some of the videos the drive is just full.
If I remember it correctly from the time I tried your image out, the partition is only around 1gb big

tofuSCHNITZEL, 2022-03-02:
@Lunal-Tic please ssh to your videolooper.de install and share the output of the command `lsblk` or if it does not work use `df -h`

tofuSCHNITZEL, 2022-03-02:
its all documented here: https://videolooper.de/
"video"

Lunal-Tic, 2022-03-02:
Thank you.  I was using the wrong password "videopi".   Here's the info:

pi@raspberrypi:~ $ lsblk
NAME        MAJ:MIN RM  SIZE RO TYPE MOUNTPOINT
mmcblk0     179:0    0 59.5G  0 disk
├─mmcblk0p1 179:1    0  256M  0 part /boot
└─mmcblk0p2 179:2    0  1.6G  0 part /

pi@raspberrypi:~ $ df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/root       1.6G  1.6G     0 100% /
devtmpfs        404M     0  404M   0% /dev
tmpfs           436M     0  436M   0% /dev/shm
tmpfs           436M  5.8M  431M   2% /run
tmpfs           5.0M     0  5.0M   0% /run/lock
tmpfs           436M     0  436M   0% /sys/fs/cgroup
tmpfs           1.0G  8.0K  1.0G   1% /tmp
tmpfs            50M   20K   50M   1% /var/log
/dev/mmcblk0p1  253M   52M  201M  21% /boot

christiansievers, 2022-03-02:
so it's really this:

> Just something that came to my mind. Maybe your image @christiansievers does not expand the partition to the whole sd card and after copying some of the videos the drive is just full. If I remember it correctly from the time I tried your image out, the partition is only around 1gb big

tofuSCHNITZEL, 2022-03-02:
yes indeed, so I remembered correctly. expanding it later should work via dietpi-config right?
@Lunal-Tic please call `sudo dietpi-config` via ssh and look in the menu there if you can find an "expand filesystem" or similar
and then run lsblk again and check if the size of "mmcblk0p2" is more than 1,6GB
@christiansievers: you you think you can include autoexpansion on first boot into you image?

christiansievers, 2022-03-02:
sounds like a plan!

Lunal-Tic, 2022-03-02:
As a follow-up as mentioned in the OP I've seen some comments about running this off the SD rather than USB drive not being ideal.  Are there any substantial cons to running off the SD fully vs video files on a USB drive?  Does one wear out faster, run hotter, etc?

tofuSCHNITZEL, 2022-03-02:
never had any issue with it so far.

tofuSCHNITZEL, 2022-03-02:
should not be a problem

closed 2022-03-02 by tofuSCHNITZEL

christiansievers, 2022-03-14:
I just uploaded a newer version of the image file. This fixes this problem by automatically expanding the file system if necessary. As a bonus it's now using v1.0.11.

Lunal-Tic, 2022-03-14:
> I just uploaded a newer version of the image file. This fixes this problem by automatically expanding the file system if necessary. As a bonus it's now using v1.0.11.

Fantastic!  Many, many thanks.  Off to pluck a couple SD cards.  :)

Lunal-Tic, 2022-03-15:
Here's the update.  v. 2.6 was installed and setup for USB copy mode on Samsung 64GB SD.  254 MP4 files (26.7GB) were copied from a Sandisk 64GB USB drive in about 40 minutes.  System was rebooted without the USB and it has been running the videos for the last 15 minutes.  

Odd thing is that during the copy, boot and first video playback I was getting low voltage warnings.  Never remember seeing that previously on the exact same hardware setup using the Legacy version of Raspberry Pi OS Lite install as recommended above.