adafruit/pi_video_looper #146: suggestion: use pip3 install instead of setuptools/easy install

christiansievers, 2022-03-15:
Going forward, I think we need to stop using setuptools to install Python modules. While omxplayer still works on an up-to-date Buster system, it seems that modules installed by setuptools aren't found any more. 

To give some background, yes, the video_looper still installs fine on the current Raspbian OS Lite Legacy version. I really like DietPi for its ease of use, robustness and easy addition of audio DACs. So I have spent quite a while to get the video_looper working on a fresh DietPi install (using a Buster image as its base), and it turns out that omxplayer is still working fine, even if you have to go through some [manual steps](https://dietpi.com/phpbb/viewtopic.php?p=37995#p37995).  

But even then the video_looper won't start, because supervisor cannot find the modules, which are installed as egg files in `/usr/lib/python3.7/site-packages`. Manually moving them to `/usr/lib/python3/dist-packages/` fixes that problem. This [issue](https://github.com/pypa/setuptools/issues/2059) here helped me solve that problem.

For reference, the install process mentions this problem:
```
SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
setuptools.SetuptoolsDeprecationWarning,
/usr/local/lib/python3.7/dist-packages/setuptools/command/easy_install.py:163: EasyInstallDeprecationWarning: easy_install command is deprecated. Use build and pip and other standards-based tools.
  EasyInstallDeprecationWarning
```

tofuSCHNITZEL, 2022-03-16:
see https://github.com/adafruit/pi_video_looper/tree/dev I will also try to install it on a fresh dietpi install

christiansievers, 2022-03-16:
ah, great. I'll also try that. DietPi don't keep around older versions of their images, so you'd need to get the last Buster image from https://web.archive.org/web/20211030184306/https://dietpi.com/downloads/images/

There's a lengthy auto-update routine that you can't stop. Good to have a screen connected for that. Once logged in as root (pw is dietpi):

In dietpi-config set the memory split to 128.

There's no pi user so you'd need to 
```
adduser pi
adduser pi sudo 
adduser pi adm
usermod -a -G video pi
apt-get install git 
reboot
```
then install it as pi (easier than to amend the looper install script I guess)

Then to get omxplayer running you need to link libraries: [OMXPLAYER on Bullseye fresh install (32-bit) - DietPi](https://dietpi.com/phpbb/viewtopic.php?p=37995#p37995). Test omxplayer on its own if you like.

Now installing via pip3 should work. Ideally. 

How much work would it be to get rid of supervisor in a next step and use systemd instead? :)

tofuSCHNITZEL, 2022-03-16:
thanks. I will check it out when I have time (will be a while)
there is already a branch for systemd:
https://github.com/tofuSCHNITZEL/pi_video_looper/tree/systemd

the issue is that there are problems with the "run-level" (is this correct?) where systemd runs the looper not in the right "session" and so it cant output to the screen.

christiansievers, 2022-03-21:
> see https://github.com/adafruit/pi_video_looper/tree/dev I will also try to install it on a fresh dietpi install

I tried that just now. Installing the script brings up this error. Also if I do `pip3 install wheel` beforehand. 

```
Building wheels for collected packages: Adafruit-Video-Looper
  Running setup.py bdist_wheel for Adafruit-Video-Looper ... error
  Complete output from command /usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-req-build-scpb19vv/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/pip-wheel-qgosgo_1 --python-tag cp37:
  usage: -c [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
     or: -c --help [cmd1 cmd2 ...]
     or: -c --help-commands
     or: -c cmd --help
  
  error: invalid command 'bdist_wheel'
  
  ----------------------------------------
  Failed building wheel for Adafruit-Video-Looper
  Running setup.py clean for Adafruit-Video-Looper
Failed to build Adafruit-Video-Looper
Installing collected packages: Adafruit-Video-Looper
  Running setup.py install for Adafruit-Video-Looper ... done
Successfully installed Adafruit-Video-Looper-1.0.10
```

The supervisor error log says `/usr/bin/python3: Error while finding module specification for 'Adafruit_Video_Looper.video_looper' (ModuleNotFoundError: No module named 'Adafruit_Video_Looper')`

tofuSCHNITZEL, 2022-03-21:
okay yes it was totaly untested and I also have 0 experience with this kind of things.. so it seems back to the drawing board...

christiansievers, 2022-03-23:
I'm struggling with committing to the dev branch of my fork (is that possible at all? what would be best practice?), so let me describe here what works for me. Seems somehow pip3 cannot properly build the wheel if the package files are all in the same Adafruit_Video_Looper folder.   

* move all files in Adafruit_Video_Looper except setup.py into a subfolder with the same name. 

```
Adafruit_Video_Looper
├── Adafruit_Video_Looper
│   ├── __init__.py
│   ├── alsa_config.py
│   ├── (etc.)
│   └── (etc.)
└── setup.py
```

* then edit this line in install.sh: `pip3 install ./Adafruit_Video_Looper` to

```
cd Adafruit_Video_Looper # pip3 needs to run from package root dir
pip3 install . # to debug use pip3 -v install . | grep 'adding' 
cd ..
```

* This then works right away in the current Raspberry OS Legacy. To get it to run on the last DietPi Buster, you also need to `apt install  python3-wheel`

tofuSCHNITZEL, 2022-04-13:
> There's no pi user so you'd need to

just thought about this... if we would place the default video directory somewhere else it would not matter if there is a pi user or not... right? should we move the video folder to a "neutral" location?