adafruit/pi_video_looper #149: using systemd instead of supervisor

christiansievers, 2022-03-24:
I had a go at the idea to run it via systemd, using the branch you started. It appears the devil is in the details of the video_looper.service file. I noticed two things. 

* `journalctl -fu video_looper` showed of module not found errors. All the info about systemd I found tells me the ExecStart command needs full paths. So just as a test I ran something simpler, and this works fine:

```
# systemd configuration to run video looper at boot and
# ensure it runs continuously.

[Unit]
Description=Adafruit Raspberry Pi Video Looper
After=getty.target
# after login prompt appears

[Service]
TimeoutStartSec=0
Type=simple
ExecStart=/usr/bin/omxplayer -r --loop -o both "/home/pi/testvideo.mp4"
#ExecStart=/usr/bin/python3 -u -m Adafruit_Video_Looper.video_looper 
Restart=always
RestartSec=3
User=root

[Install]
WantedBy=multi-user.target

```

So the error lies with `ExecStart=sudo /usr/bin/python3 -u -m Adafruit_Video_Looper.video_looper`. Probably the 2nd half of that line. The examples I found don't use python packages but single python script files. I don't know how to link to the package. `/usr/local/lib/python3.7/dist-packages/Adafruit_Video_Looper.video_looper` doesn't seem to work. And it may install somewhere else on another system.

* Also note the `After=getty.target` which tells systemd to run it after the login prompt appears. 

Just sharing my notes with you, no rush...

tofuSCHNITZEL, 2022-04-21:
just a heads up. this is already part of the dev branch (together with the pip install change which solves the module not found error) still needs some testing tho

tofuSCHNITZEL, 2022-04-25:
@christiansievers: I'm quite far regarding this change... the only issue is that running omxplayer as non root - (which is mandatory with cvlc - so better to figure out now) is that the image from the omxplayer process is just not visible... maybe you have some ideas and could try out the dev branch...
I also opted to try to "harmonise" the install across different PI OSs by adding the pi user if non existent so the player can run under that user.

tofuSCHNITZEL, 2022-04-27:
yeah sorry one thing led to another