adafruit/pi_video_looper #150: VLC support

JoakimBoSWE, 2022-04-04:
Hello,
I am using your latest version to play regular MP4 files, in a loop.
Using a Raspberry Pi 4 model b, connected with micro-hdmi cable.
Works like a charm, however they are with audio on the computer, but when play thru the Pi video looper, i dont get any sound out thru the HDMI.
Is there any settings that sets audio out port for the HDMI?

Also will there be any updates to support VLC in the upcoming releases?
Thx :)

tofuSCHNITZEL, 2022-04-11:
duplicate of  #124

tofuSCHNITZEL, 2022-04-11:
please check /boot/config.txt there is a setting:
```
# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2
```
maybe this helps you. for me audio via hdmi is working just fine with rpi4 and Legacy version of Raspberry Pi OS Lite

vlc support ist comming Soon™ see https://github.com/adafruit/pi_video_looper/issues/124

tofuSCHNITZEL, 2022-10-03:
The pb_bullseye branch is WIP and alpha at best. But feel free to contribute.

closed 2023-11-16 by tofuSCHNITZEL