adafruit/pi_video_looper #152: Login?

RyanWoirol, 2022-04-16:
I had Video Looper set up on a Pi Zero and a 3.5in tft from Waveshare.  Everything was working great, but I wanted to rotate the orientation.  Now every time I try to run it, I get a command screen that says 

"Raspbian GNU/Linux 10raspberrypi ttyl2
raspberrypi login:"

and there is a flashing cursor.

Trying to hook up a keyboard and it doesn't let me type.

tofuSCHNITZEL, 2022-04-16:
Try to connect via ssh and see if you can do things there.
I noticed a similar thing over ssh, if you start the looper via the ssh session it will not display what you type anymore. Maybe this happens to the regular terminal as well...

RyanWoirol, 2022-04-17:
I can still SSH in.  I tried rotating the screen back, thinking maybe that would undo it and no dice.  I've tried to startx and it will load up Raspbian, but the video looper never loads.

RyanWoirol, 2022-04-17:
I take that back.  I can no longer startx.  I get this

pi@raspberrypi:~ $ startx

X.Org X Server 1.20.4
X Protocol Version 11, Revision 0
Build Operating System: Linux 5.4.0-72-generic armv8l Raspbian
Current Operating System: Linux raspberrypi 5.10.103+ #1529 Tue Mar 8 12:19:18 GMT 2022 armv6l
Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=480 bcm2708_fb.fbheight=320 bcm2708_fb.fbswap=1 smsc95xx.macaddr=B8:27:EB:E4:EA:EA vc_mem.mem_base=0x1ec00000 vc_mem.mem_size=0x20000000  console=ttyS0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet splash fbcon=map:10 fbcon=font:ProFont6x11
Build Date: 04 January 2022  10:05:34AM
xorg-server 2:1.20.4-1+rpt4+deb10u4 (https://www.debian.org/support)
Current version of pixman: 0.36.0
        Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
(==) Log file: "/home/pi/.local/share/xorg/Xorg.1.log", Time: Thu Apr 14 05:41:59 2022
(==) Using system config directory "/usr/share/X11/xorg.conf.d"
(EE)
Fatal server error:
(EE) parse_vt_settings: Cannot open /dev/tty0 (Permission denied)
(EE)
(EE)
Please consult the The X.Org Foundation support
         at http://wiki.x.org
 for help.
(EE) Please also check the log file at "/home/pi/.local/share/xorg/Xorg.1.log" for additional information.
(EE)
(EE) Server terminated with error (1). Closing log file.
xinit: giving up
xinit: unable to connect to X server: Connection refused
xinit: server error
Couldn't get a file descriptor referring to the console

tofuSCHNITZEL, 2022-04-17:
Okay, how did you rotate the screen?
What's in the video looper (supervisor) logs?

closed 2022-05-16 by tofuSCHNITZEL