adafruit/pi_video_looper #154: Problem with new version of python3.9

tygroo, 2022-07-16:
Hello,
I updated my raspberry pi and since then I have problems launching video_looper.
Indeed, it seems that python 3 has been updated and that since then, the pygame library no longer works as expected.

I have the error:
"
Current thread 0x76f0b7e0 (most recent call first):
   File "/usr/local/lib/python3.9/dist-packages/Adafruit_Video_Looper-1.0.11-py3.9.egg/Adafruit_Video_Looper/video_looper.py", line 78 in __init__
   File "/usr/local/lib/python3.9/dist-packages/Adafruit_Video_Looper-1.0.11-py3.9.egg/Adafruit_Video_Looper/video_looper.py", line 471 in <module>
   File "/usr/lib/python3.9/runpy.py", line 87 in _run_code
   File "/usr/lib/python3.9/runpy.py", line 197 in _run_module_as_main
"
There seems to be a problem with pygame.
python 3.9.2
pygame 1.9.6+dfsg-4+b1
How can I fix the situation?
Thank you for your help.

tofuSCHNITZEL, 2022-07-16:
Maybe try to update pygame to the latest version.
`pip3 install --upgrade pygame`

tygroo, 2022-07-17:
Hello,
I tried to install this way with and without "sudo" but I still have the problem.
Here are the logs from the stdout file:
`pygame 2.1.2 (SDL 2.0.14, Python 3.9.2)
Hello from the pygame community. https://www.pygame.org/contribute.html
Starting Adafruit Video Looper.
The path /dev/dri/ cannot be opened or is not available
The path /dev/dri/ cannot be opened or is not available
`
And the logs of stderr file : 
`
Fatal Python error: pygame_parachute: (pygame parachute) Segmentation Fault
Python runtime state: initialized

Current thread 0x76d0b7e0 (most recent call first):
  File "/usr/local/lib/python3.9/dist-packages/Adafruit_Video_Looper-1.0.11-py3.9.egg/Adafruit_Video_Looper/video_looper.py", line 78 in __init__
  File "/usr/local/lib/python3.9/dist-packages/Adafruit_Video_Looper-1.0.11-py3.9.egg/Adafruit_Video_Looper/video_looper.py", line 471 in <module>
  File "/usr/lib/python3.9/runpy.py", line 87 in _run_code
  File "/usr/lib/python3.9/runpy.py", line 197 in _run_module_as_main
`
I'm on a raspberry pi.
uname -a => Linux raspipetitetv 5.15.53-v7+ #1572 SMP Wed Jul 13 17:13:02 BST 2022 armv7l GNU/Linux

What I understand is that I lack a video output interface.
However, the hdmi port is functional.

tofuSCHNITZEL, 2022-07-17:
Did you install the Legacy version of Raspberry Pi OS Lite?

tygroo, 2022-07-22:
Hello,
Yes I installed the legacy version of raspberry pi os lite.
The problem happened after an `sudo apt update` then an `sudo apt upgrade`.
I have since changed my sourcelist to upgrade to bullseye. 
But I have the same problems.
Maybe I should downgrade my version of python3 (and pygame because it's the problem) but I'm not sure how to proceed.

tygroo, 2022-07-27:
Hello,
Finally, despite the fact that I didn't want to physically touch the raspberries, I reinstalled a "legacy" buster version of Raspberry pi OS.
Everything works as before. And I will no longer update my raspberry.
Thank you for your help.

closed 2022-07-27 by tygroo