adafruit/pi_video_looper #155: Display ticking time for 5 seconds between videos

DanBaigo, 2022-08-30:
Hello, I'm looking to try and display a nice clear time display in the middle of the screen for 5 seconds between videos. I thought maybe I could do something with the bgimage = display somehow.

tofuSCHNITZEL, 2022-08-30:
what exactly do you want to display? the current time?

tofuSCHNITZEL, 2022-08-31:
okay, this can be done fairly easy with appropriate pygame code - but please keep in mind that the rpi would need to be connected to the internet and have NTP enabled because it is missing a real time clock and will not be able to keep the time very accurately...

DanBaigo, 2022-08-31:
Hi Tofu, could you please help me with this script? I am not sure how to write this code.

tofuSCHNITZEL, 2022-09-01:
But are you prepared to connect and setup ntp on the pi or connect a RTC module?
Otherwise displaying the time just does not make any sense (bcs it will be wrong)

DanBaigo, 2022-09-01:
Yes I can connect and setup NTP no problem.

Just not sure on the coding to display time.

tofuSCHNITZEL, 2022-09-03:
I have implemented it in this branch:
https://github.com/adafruit/pi_video_looper/tree/feat_timedisplay

you need to enable it in the video_looper.ini (and set a wait time)

DanBaigo, 2022-09-03:
Thanks so much! That works perfectly!

I have been playing around with the format codes that you included (
https://docs.python.org/3/library/datetime.html#strftime-and-strptime-format-codes)
but when I added a few extra bits in the text ended up going off the screen.

If I wanted to put Day of week, date, month, year on line 1 and then H:M:S
on line 2 is that simple to add in? Or is there a slightly different way
the "timeLabel =
self._render_text(now.strftime(self._datetime_display_format),
self._big_font)" line would need to be modded?

Have been faffing for hours and can't get it to work. Seems so simple.

Thought it would just look something like
%A %d %B %Y
then newline= or \n (not sure what the command is for a new line)
%H:%M:%S

Thanks for all your help,

Dan Baker

tofuSCHNITZEL, 2022-09-04:
You just need to look a couple of lines above, there you will find the code for the countdown which is with two lines.
Pygame does not render new line characters so you need to render a second line of text.
So you need to duplicate also the size vars etc. Exactly like it is with the code above for the countdown.

DanBaigo, 2022-09-07:
Hi Tofu, I still can't get this to work. Tried editing the code live on the PI, in the .py but its not changing the running code. I have tried updating the code base so I can resend it to my pi but I can't because the directory already exists, I can't figure out how to uninstall so I can start fresh, I don't want to wipe whats already on the PI. I have no idea what I'm doing.

Could you please put me out of my misery with this one. I'd be eternally grateful.

tofuSCHNITZEL, 2022-09-07:
See the install.sh file... After you Change the source you need to install the module again so it uses the modified code.

DanBaigo, 2022-09-08:
Makes sense, no wonder nothing was changing. I’m now doing sudo
./install.sh between changes.

So I added in the extra lines just like you said and changed nothing else
just to see what would happen and then did a sudo ./install.sh. (Both
labels read something like self._date_display_format which were identical).

This gave me 2 sets of time between clips, the top one static and the
bottom one dynamically incrementing but 2 lines which is great, felt like I
was starting to make some progress.

Then I thought all I’d need to do was essentially make each line unique in
the .py and add an extra line into the .ini targetting that same new unique
line.

Now the program doesn’t start at all and just crashes since I’ve done that.

The below is the code I’ve got set up now, any idea where I have gone wrong?

video_looper.py settings

def _display_datetime(self):

    sw, sh = self._screen.get_size()

    for i in range(self._wait_time):

        now = datetime.now()

        timelabel1 =
self._render_text(now.strftime(self._date_display_format), self._big_font)

    l1w, l1h = timelabel1.get_size()

    sw, sh = self._screen.get_size()

    for i in range(self._wait_time):

        timelabel2 =
self._render_text(now.strftime(self._time_display_format), self._big_font)

        l2w, l2h = timelabel2.get_size()

        # Clear screen and draw text with line1 above line2 and all

        # centered horizontally and vertically.

        self._screen.fill(self._bgcolor)

        self._screen.blit(timelabel1, (round(sw / 2 - l1w / 2), round(sh /
2 - l2h / 2 - l1h)))

        self._screen.blit(timelabel2, (round(sw / 2 - l2w / 2), round(sh /
2 - l2h / 2)))

        pygame.display.update()

        # Pause for a second between each frame.

        time.sleep(1)

.ini file settings

datetime_display = true

# this controls the format the date/time is displayed: (see
https://docs.python.org/3/library/datetime.html#strftime-and-strptime-format-codes
)

date_display_format = %H:%M:%S

time_display_format = %A %d %B %Y

tofuSCHNITZEL, 2022-09-09:
this is not a good way to share code. please use the format as code option here - or better yet, since you made a fork anyway... commit to your fork and I can have a look there. 

you dont need to run the full install.sh - `python3 setup.py install --force` is enough.

to see the logs and py error messages and then debug your code use: `Use sudo tail -f /var/log/supervisor/video_looper-stdout* and sudo tail -f /var/log/supervisor/video_looper-stderr* to view the logs.`

DanBaigo, 2022-09-15:
Thanks, I have committed these changes to my fork.

Could you please take a look and fix? Would be very grateful.

tofuSCHNITZEL, 2022-09-20:
Did you check the logs...? the error messages there should give you an idea of what is wrong... (hint: you are referencing variables that don't exist e.g. `self._time_display_format` and indentation of your function `_display_datetime` is wrong)

tofuSCHNITZEL, 2022-09-20:
Did you check the logs...? the error messages there should give you an idea of what is wrong... (hint: you are referencing variables that don't exist e.g. `self._time_display_format` and indentation of your function `_display_datetime` is wrong)

sorry but this is here is not a python course - thank you for the suggestion of this feature - a basic version is implemented and released and can be improved upon - so I will close this issue for now - I hope you understand.

closed 2022-09-20 by tofuSCHNITZEL

DanBaigo, 2022-10-11:
Hi, makes sense. How would I go about this?

I'm very novice when it comes to coding.

DanBaigo, 2022-10-11:
Appreciate this seems frustratingly simple but I've spent hours on this now
and I still can't get it to work.

DanBaigo, 2023-02-16:
I have now implemented the additional feature requested.