adafruit/pi_video_looper #156: Video looper running ok for months but now freezing

seanocaster123, 2022-09-01:
I have an install of adafruit video looper running on a pi4 with 2GB of ram and a 16GB sanddisk sd card. The whole setup has been playing 1920x1080 videos perfectly fine via HDMI to a TV on loop for maybe 10 months. Within the last few days it has been freezing and I'm trying to trouble shoot. No software has been updated...

I'm no pro but from what I can tell using htop is that the RAM slowly fills up and then the swap file slowly fills up - over a period of about 12 hours or more. Eventually the pi locks up and I can't ssh in (because it's at a remote location). If I stop the player, turn swapoff and then back on again I see all the RAM empty. I don't understand how a player that has been working fine suddenly develops this problem.

Has anyone seen this before?

christiansievers, 2022-09-01:
It might be hardware failure. SD cards wear out, especially when it’s hot. That depends on luck and on the quality of the card. And I’ve seen power adapters breaking (though only after some years of continuous service).

seanocaster123, 2022-09-01:
Thanks for the insight. The pi has been running continuously since Feb 2021 so I guess it's likely that the sd card is giving up. I will try and replace it and then give an update. Thank you.

tofuSCHNITZEL, 2023-11-16:
Stale

closed 2023-11-16 by tofuSCHNITZEL