adafruit/pi_video_looper #163: Pause vs stop

Chrisgraue, 2022-11-05:
Pause (S key) doesn't work as expected.

Pressing S stops video playback, turning screen black. When S is pressed again, the next video in the playlist starts from the beginning.

Is it possible to have the same video file resume from the moment it was paused? Like how a pause/play button normally works on most devices?

tofuSCHNITZEL, 2022-11-05:
Hi,
the documentation states that S is stop/start and not pause. Unfortunately the way the video looper is currently designed, there is no way to send "signals" to the software (omxplayer) that actually plays the videos to make them halt or resume.
But I can have a look to change the behaviour so that pressing S again starts the same video from the beginning where S was pressed the first time...

Thanks for understanding.
Cheers

Redacted99, 2023-02-06:
Sending a command to omxplayer is pretty easy.  I have coded it for my application and can add it to the software if you approve.  Basically add stdin=subprocess.PIPE, to the popen and then add a pause function to omxplayer.py that sends a space character to omxplayer when you want to pause it.  A few lines of code in omxplayer.py and then keystroke glue in video_looper

tofuSCHNITZEL, 2023-02-06:
Great, please open a pull request!

cinghialino, 2023-03-13:
Oh I would love this too! I made a compilation of video for my aging father, so the ability to pause a movie and then resume it at a later time would be amazing.

tofuSCHNITZEL, 2023-03-14:
it will come in version 1.0.13 which is currently in the "next" branch and needs to be tested before it is "released"

tofuSCHNITZEL, 2023-11-16:
pause is implemented in https://github.com/adafruit/pi_video_looper/releases/tag/v1.0.13

closed 2023-11-16 by tofuSCHNITZEL