adafruit/pi_video_looper #164: Intro&outro for random playlist

postnuka, 2022-11-06:
Hi, 
Is it possible to add intro and outro videos to a randomly generated playlist? It will be like this:
1- intro
2- random video playlist
3- outro

Thank you.

tofuSCHNITZEL, 2022-11-06:
Not directly. You could create a playlist manually... But then the videos in the middle would have the same order every time... To have a truly random list of videos in the middle of two fixed videos, you would need to adapt the python code.

postnuka, 2022-11-06:
Thank you. I see that there is a sample playlist file called “example.m3u”. So maybe I can create a shell script for modifying this file for this purpose. A little help would be appreciated. :)

tofuSCHNITZEL, 2022-11-06:
Yes that would be a possibility but it depends when do you need the randomness to be "rerolled"... Because your shell script would need to restart the video looper so it reads the m3u file again. But there is no feedback from the player when it reached the outro file...

postnuka, 2022-11-06:
Yeah, I got it. But I think it's good enough.
I would be very grateful if you can recommend a reference I can follow about these shell script.

tofuSCHNITZEL, 2022-11-06:
Maybe tell me more about what you need exactly then we will see if it is even possible with a shell script.

closed 2023-11-16 by tofuSCHNITZEL