adafruit/pi_video_looper #166: Running on the Libre Potato

iamdoctorbrain, 2022-11-09:
Has anyone had success running this on on the Libre Potato?  I don't have the skills to install it myself.  I was trying to see if someone created a standalone plug and play SD image like the one that exists for the Raspberry Pi?  Would use the Raspberry Pi but impossible to find at a reasonable price right now.

Thanks

christiansievers, 2022-11-10:
It’s a good question, now that Raspberries are hard to get: Will the looper run on other single board computers? 

My concern would be that not all, if any, have the Pi‘s dedicated video playback chip, which enables this comparatively weak hardware to smoothly play 1080p h.264 video.

christiansievers, 2022-11-18:
here's someone's write-up on how to get Raspi OS run on Libre computers: https://jamesachambers.com/enable-raspbian-images-to-boot-on-libre-computers-boards/
Let's see what he says about 1080p playback...

christiansievers, 2022-11-19:
these SBCs actually have powerful GPUs so they *should* be able to play back 1080p, [he says](https://jamesachambers.com/enable-raspbian-images-to-boot-on-libre-computers-boards/#comment-15001).  @iamdoctorbrain did you have a go yet?

christiansievers, 2023-01-03:
I finally got my hands on a Libre "potato", but it seems omxplayer is Raspberry Pi specific and uses Raspberry Pi specific hardware. We'll have to wait for the VLC branch of this script to become ready I guess...
@tofuSCHNITZEL please close

closed 2023-01-03 by tofuSCHNITZEL