adafruit/pi_video_looper #168: Attempting to add ASLA volume controls

Pnub, 2022-12-09:
Trying to add new keyboard shortcuts to allow for volume up and volume down, but any changes made to the video_looper.py script aren't reflected in the restart.  Even a simple additional output in run to the console.  What am I missing?  

Once I get this working I'm curious if others would find it worthwhile.

SirVaxi, 2023-01-21:
Hey Pnub 
Did you succeed? - im trying the same thing, but i can not get it to work. If you did, will you share the code ?

Pnub, 2023-01-30:
SirVaxi,
I was successful.  Let me make a few tweaks and I'll upload it to my repo to share.

tofuSCHNITZEL, 2023-01-30:
great! please open a pull request after you are done, thanks!

SirVaxi, 2023-01-31:
Pnub,
Looking forward to seeing how you solved it! - I think it's so nice that you want to share it with the rest of us!

tofuSCHNITZEL, 2023-11-17:
maybe it would also work to just send "+" and "-" to the omxplayer process? (just like the pause command)
needs to be tested if this would also control the alsa device