adafruit/pi_video_looper #170: Video looper stops working after reboot

inomie, 2023-01-20:
I have used the video looper on 5 raspberry pi's and for all of them it works. Now I'm setting this up on the 6th one and its not working. After i have run the command "sudo ./install.sh" the omxplayer starts. When I restarting the pi it's not starting again. I have tried with one of the other's sd cards and it works fine. I have also reinstalled the pi and done the process once more but the same problem. I have even tried with another sd card but that's the same problem. 

When I try to run the command "python3 -u -m Adafruit_Video_Looper.video_looper" The screen turns black and then goes back to the desktop. The terminal say's "AttributeError: module "pyudev" has no attribute 'Context'"

tofuSCHNITZEL, 2023-01-20:
running pi_video_looper on any other installation than: (Raspberry Pi OS Lite (Legacy))
https://www.raspberrypi.com/software/operating-systems/#raspberry-pi-os-legacy
LITE! (without desktop)
is not supported

inomie, 2023-01-20:
The other raspberry pi's are running on Raspberry pi OS legacy. So it does not need to be lite.

tofuSCHNITZEL, 2023-01-20:
yes, sure it can work. but its not tested with it and therefore not supported... but If you post the log output maybe I can help anyway... also please share the whole terminal output of "python3 -u -m Adafruit_Video_Looper.video_looper"

inomie, 2023-01-20:
pi@raspberrypi:~/pi_video_looper $ python3 -u -m Adafruit_Video_Looper.video_looper
pygame 1.9.4.post1
Hello from the pygame community. https://www.pygame.org/contribute.html
Starting Adafruit Video Looper.
Traceback (most recent call last):
  File "/usr/lib/python3.7/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/pi/pi_video_looper/Adafruit_Video_Looper/video_looper.py", line 495, in <module>
    videolooper = VideoLooper(config_path)
  File "/home/pi/pi_video_looper/Adafruit_Video_Looper/video_looper.py", line 87, in __init__
    self._reader = self._load_file_reader()
  File "/home/pi/pi_video_looper/Adafruit_Video_Looper/video_looper.py", line 128, in _load_file_reader
    return importlib.import_module('.' + module, 'Adafruit_Video_Looper').create_file_reader(self._config, self._screen)
  File "/home/pi/pi_video_looper/Adafruit_Video_Looper/usb_drive.py", line 46, in create_file_reader
    return USBDriveReader(config)
  File "/home/pi/pi_video_looper/Adafruit_Video_Looper/usb_drive.py", line 18, in __init__
    readonly=self._readonly)
  File "/home/pi/pi_video_looper/Adafruit_Video_Looper/usb_drive_mounter.py", line 23, in __init__
    self._context = pyudev.Context()
AttributeError: module 'pyudev' has no attribute 'Context'

inomie, 2023-01-20:
When im gonna check the logs it says that no such file or directory

tofuSCHNITZEL, 2023-01-20:
Videolooper needs to run as root, otherwise it can't mount the USB drive. The thing you posted above shows that you did not run it as root.

Also please check if /mnt/usbdrive0 even exists (did you run install.sh as root?)

inomie, 2023-01-20:
okey, when I did the Os setup it's just logging me in automatic

inomie, 2023-01-20:
Okey, it's say's exactly the same

tofuSCHNITZEL, 2023-01-20:
Okay does /mnt/usbdrive0 exist?
Do you see your usb drive when running `lsblk`? 
Can you try to manually mount the stick to the above location. 

What's the output of `python3 -m pip show pyudev`

inomie, 2023-01-20:
1. yes/mnt/usbdrive0 exist.
2. Yes I can see it.
3. output: 
Name: pyudev
Version: 0.24.0
Summary: None
Home-page: None
Author: None
Author-email: None
License: None
Location: /usr/local/lib/python3.7/dist-packages/pyudev-0.24.0-py3.7.egg
Requires: 
Required-by: Adafruit-Video-Looper

tofuSCHNITZEL, 2023-01-20:
Hm not really sure what the problem ist. I have the feeling it's obvious, but I don't really see it... I can only recommend that you try to start over, Flash a fresh LITE os and install it according to the readme...

christiansievers, 2023-01-20:
Try swapping Raspis and power supplies? Any of those could be bad and causing strange problems. 

Also, clone a working SD card onto a problematic one. If it still acts up, it’s the card, I would say.

closed 2023-11-16 by tofuSCHNITZEL