adafruit/pi_video_looper #172: 16:10 aspect ratio videos distorted on rpi 3b+

alreadyconfused, 2023-01-31:
Hi! I've been trying to get a 1920x1200 video working on the videolooper that I could use on a projector, but the bottom of the video gets all distorted. I tried using different formats on omxplayer, also tried hello_video but it still had the same issue.
[image]

JonnyTech, 2023-01-31:
Does it work with a display resolution of 1920x1080?

alreadyconfused, 2023-01-31:
Yeah it does, but the only problem is that the projector has a 16:10 aspect ratio. If I try to play a 16:9 video using hello_video, white bars show up in the top and bottom. Omxplayer solves the white bars, but then the loop isn't as seamless. Not a huge issue, but just wanted to inform that 16:10 might not work

JonnyTech, 2023-01-31:
Are you sure that the resolution of the video is  actually 1920x1200? I suspect that it is 1920x1080 and your display is 1920x1200 - afaik pi_video_looper does not rescale.

alreadyconfused, 2023-01-31:
Yeah 100% sure. I changed it in premiere, then used ffmpeg to make the mp4 into a h264 file.

christiansievers, 2023-01-31:
I think the video file’s max video resolution might be 1920x1080. Something to do with the dedicated playback chip.

alreadyconfused, 2023-01-31:
Alright yeah I was wondering if it's just a hardware limitation on my case. Thanks for the responses!

closed 2023-01-31 by alreadyconfused