adafruit/pi_video_looper #177: After moving from Micro SD card to SSD, the looper no longer starts

jlindemeier, 2023-03-12:
The video looper works great with a micro SD card.
After moving to an SSD (due to reliability), the looper no longer starts (exit status 32).
Maybe someone has an explanation or solution.
Thanks!

sudo tail -f /var/log/supervisor/video_looper-stdout*
=====================================================
Starting Adafruit Video Looper.
pygame 1.9.4.post1
Hello from the pygame community. https://www.pygame.org/contribute.html

sudo tail -f /var/log/supervisor/video_looper-stderr*
=====================================================
   exec(code, run_globals)
  File "/usr/local/lib/python3.7/dist-packages/Adafruit_Video_Looper-1.0.12-py3.7.egg/Adafruit_Video_Looper/video_looper.py", line 500, in <module>
  File "/usr/local/lib/python3.7/dist-packages/Adafruit_Video_Looper-1.0.12-py3.7.egg/Adafruit_Video_Looper/video_looper.py", line 406, in run
  File "/usr/local/lib/python3.7/dist-packages/Adafruit_Video_Looper-1.0.12-py3.7.egg/Adafruit_Video_Looper/video_looper.py", line 209, in _build_playlist
  File "/usr/local/lib/python3.7/dist-packages/Adafruit_Video_Looper-1.0.12-py3.7.egg/Adafruit_Video_Looper/video_looper.py", line 218, in _build_playlist_from_all_files
  File "/usr/local/lib/python3.7/dist-packages/Adafruit_Video_Looper-1.0.12-py3.7.egg/Adafruit_Video_Looper/usb_drive.py", line 30, in search_paths
  File "/usr/local/lib/python3.7/dist-packages/Adafruit_Video_Looper-1.0.12-py3.7.egg/Adafruit_Video_Looper/usb_drive_mounter.py", line 47, in mount_all
  File "/usr/lib/python3.7/subprocess.py", line 347, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['mount', '-r', '/dev/sda1', '/mnt/usbdrive0']' returned non-zero exit status 32.

lsb_release -a
==============
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 10 (buster)
Release:        10
Codename:       buster

sudo mount -r /dev/sda1 /mnt/usbdrive0
======================================
mount: /mnt/usbdrive0: /dev/sda1 already mounted on /boot.

echo $?
=======
32

tofuSCHNITZEL, 2023-03-12:
Hi,
it seems that you are using usb_drive or usb_drive_copymode and the looper can not distinguish between a usb stick and your external SSD, so it is trying to mount it for playing videos from it which fails.
So currently you can only use "directory" mode if you want to boot from an external ssd.
cheers.

jlindemeier, 2023-03-13:
Hello,
thank you very much for your quick answer!
I will use the looper in the "directory" mode.
Best regards!

closed 2023-03-13 by jlindemeier