adafruit/pi_video_looper #179: Video Looper for universal linux installation / remove omxplayer dependency

alexschultze, 2023-03-20:
Video looper is a great tool, but the prices for PI's are still very high.
There is not much missing to make Video Looper a universal tool, running under Debian / Ubuntu.

I have tried to make it install and run using a recent Ubuntu LTS. For installation, i have removed the omxplayer dependency, then it install fine.
Trying to run, it recognizes the USB files fine. It shows the countdown, but then enters the Desktop environment.

tofuSCHNITZEL, 2023-03-20:
If we can get cvlc to work (and also system) then it will probably work with any platform. There is already a branch for cvlc support but it's not that easy to get all of the stuff working (cvlc wants non root / but usb mount needs root (or needs to be replaced by something else that mounts the drives) / and finally when using sydtemd and nonroot somehow it does not display any video because its outputting to the wrong terminal (I think)

tofuSCHNITZEL, 2023-04-13:
for now you can just install the legacy lite version

mkurss, 2023-04-13:
Thank you for the quick response. Got it working for video. Would you happen to know any similar (boots up, reads attached usb, autoplays) for mp3 or wav files? Sorry if this is not the right place for this question.

christiansievers, 2023-04-13:
#187 is roughly related

tofuSCHNITZEL, 2023-11-16:
As a workaround I would suggest putting the audio in a mp4 (black) video - so you can still use the video looper for playback ;)

closed 2023-11-16 by tofuSCHNITZEL