adafruit/pi_video_looper #182: Skip to next video on the playlist via terminal

foilandwater, 2023-04-02:
Hi,

Is it possible to skip to the next video via a terminal command? ie. the equivalent of the keyboard shortcut 'K' but done via command line.

I'm asking because the TV is mounted high up and we have no keyboard connected. The player loops 33 videos and occasionally I want to shop someone a specific video, so I was hoping to just skip and cycle through until the one I want.

Thanks in advance.

tofuSCHNITZEL, 2023-04-02:
Maybe this can work, have not tried it yet:
https://unix.stackexchange.com/questions/267704/command-to-simulate-keyboard-input

foilandwater, 2023-04-02:
Thanks, I found that. I've been fiddling but can't get it to work. I eventually found the below line of code which doesn't do anything. Is it perhaps because I'm targeting the wrong display?
`export XAUTHORITY=/home/pi/.Xauthority; export DISPLAY=:0; xdotool key k`

FYI, running only `xdotool key k` results in an error which is why i ended up with the above line of code.

tofuSCHNITZEL, 2023-04-02:
Yes I guess so, but I would have no idea which display to target

foilandwater, 2023-04-03:
Ok thanks for the reply and for the info. No worries I'll keep fiddling and will report back if I have success!

tofuSCHNITZEL, 2023-11-16:
Stale

closed 2023-11-16 by tofuSCHNITZEL