adafruit/pi_video_looper #183: Video player not randomising well

foilandwater, 2023-04-02:
Hi,

I updated the video player to try and improve the randomisation of the playlist. After the update, it seems like its still not randomising very well. You can see from the log file output below that out of 33 videos in the playlist, the first video played twice, then the third video played twice. Out of 6 plays, there were 2 instances were it was duplicated. I'm just wondering if I'm doing something wrong?

Appreciate any suggestions, thanks.

Starting Adafruit Video Looper.
[2023-04-02 08:56:12.997462] Using /home/pi/pictures/bg.png as a background
[2023-04-02 08:56:13.934425] Found 33 media files.
[2023-04-02 08:56:14.293677] Playing movie: /home/pi/videos/2023-03-17 Rip Curl Sunset Open 2023.mp4 (2023-03-17 Rip Curl Sunset Open 2023) 1/1
[2023-04-02 09:02:45.052776] Waiting for: 5 seconds
[2023-04-02 09:02:50.055435] Playing movie: /home/pi/videos/2023-03-17 Rip Curl Sunset Open 2023.mp4 (2023-03-17 Rip Curl Sunset Open 2023) 1/1
[2023-04-02 09:09:06.999411] Waiting for: 5 seconds
[2023-04-02 09:09:12.005010] Playing movie: /home/pi/videos/2020-10-06 KBC Kids Surf Lessons.mp4 (2020-10-06 KBC Kids Surf Lessons) 1/1
[2023-04-02 09:11:19.207523] Waiting for: 5 seconds
[2023-04-02 09:11:24.208797] Playing movie: /home/pi/videos/2022-03-31 The Ripcurl Sunset Open.mp4 (2022-03-31 The Ripcurl Sunset Open) 1/1
[2023-04-02 09:17:29.545655] Waiting for: 5 seconds
[2023-04-02 09:17:34.552942] Playing movie: /home/pi/videos/2022-03-31 The Ripcurl Sunset Open.mp4 (2022-03-31 The Ripcurl Sunset Open) 1/1
[2023-04-02 09:23:39.910646] Waiting for: 5 seconds
[2023-04-02 09:23:44.913029] Playing movie: /home/pi/videos/2022-05-04 KBC Wing.mp4 (2022-05-04 KBC Wing) 1/1

tofuSCHNITZEL, 2023-04-02:
The random function will pick any item of the playlist and not one of the not played items - so I would argue that even playing the same file every time would still be random (though not very likely)
So you would rather look for a randomisation that only picks not played files?

tofuSCHNITZEL, 2023-11-16:
Stale

closed 2023-11-16 by tofuSCHNITZEL