adafruit/pi_video_looper #185: GPIO Display shows .264 files with hello_video.bin but not with video looper

JarettSutula, 2023-04-07:
Hello,
I am running a Raspberry Pi 4 Model B 4GB with the Buster Legacy OS. My GPIO Display is a Waveshare 4inch DPI LCD (C) screen that plugs right into the GPIO interface.

The screen works and (normally) mirrors the same desktop work as the HDMI output on a monitor. Upon boot, the video looper countdown appears on the HDMI output (but not the GPIO display), it recognizes my file in the directory I set it as, but then on both screens it just sits at a black screen.

Running 'hello_video.bin --loop xxx.h264' in the directory at first looked like it wasn't working, but it actually does, just ONLY on the GPIO Display. It works for both unlooped and looped h264 files, and it works with test videos downloaded from the internet. Obviously something is up here - I'm assuming the hello_world.bin is recognizing my GPIO display as its main when on my RPI desktop, but for some reason the video looper itself recognizes the HDMI output as the main display but cannot seem to get the video appearing there.

I read through some of the issues and saw #6, applied that fix, and am still having the issue. If anyone could point me in the right direction that would be awesome - please let me know what else I can provide to help solve this.

tofuSCHNITZEL, 2023-04-07:
unfortunately I have no experience with the video_looper and non HDMI screens. 
does it make a difference if you run run hello_video.bin manually as root or as pi user?
maybe you need to switch the KMS:
in rpi config what is the "dtoverlay=" set to?
maybe try using `dtoverlay=vc4-fkms-v3d`

cheers

JarettSutula, 2023-04-07:
Thanks for the quick response - I really appreciate your help.

The dtoverlay bit you sent is already in the config.txt. It appears that, for the screen to work, I will have to set dtoverlay to a bunch of different things, which I am assuming is the cause of the issue.

Display wiki says to put this in config.txt for Bullseye:
```dtoverlay=vc4-kms-v3d
dtoverlay=vc4-kms-DPI-4inch
dtoverlay=waveshare-4dpic-3b-4b
dtoverlay=waveshare-4dpic-3b
dtoverlay=waveshare-4dpic-4b
```
I attempted this one (even though I am on Buster) and I actually did get the countdown text on the screen, but the video did not appear on the monitor or the screen (and the colors were sort of messed up).

The wiki for the screen said to put this in config.txt for Buster:
```gpio=0-9=a2
gpio=12-17=a2
gpio=20-25=a2
dtoverlay=dpi24
enable_dpi_lcd=1
display_default_lcd=1
extra_transpose_buffer=2
dpi_group=2
dpi_mode=87
dpi_output_format=0x7f216
dpi_timings=720 0 46 2 42 720 0 16 2 18 0 0 0 60 0 60000000 6
dtoverlay=waveshare-4dpic-3b-4b
dtoverlay=waveshare-4dpic-3b
dtoverlay=waveshare-4dpic-4b
```
I was originally working with that and it does show the desktop fine but again no sign of the video or the looper countdown with it.

Disabling the Buster code above in my config.txt actually made the video appear on my monitor! It gives me some relief that the looper is running fine and it is likely just some sort of display issue and how the RPi sets up its displays. I suppose the hard part now is getting the gpio display to mirror exactly what the HDMI output is without messing anything up too bad.

Thanks again for your help.

JarettSutula, 2023-04-07:
Ah, in hindsight it is pretty simple - Disabling the `dtoverlay=vc4-fkms-v3d` and having the only dtoverlays being the ones in the Buster branch setup works flawlessly! The Pi no longer has HDMI output but that's fine for what I'm doing. 
Thank you again!

closed 2023-04-07 by JarettSutula

tofuSCHNITZEL, 2023-04-07:
Glad that it worked!

mo-nonym, 2023-09-16:
> Ah, in hindsight it is pretty simple - Disabling the `dtoverlay=vc4-fkms-v3d` and having the only dtoverlays being the ones in the Buster branch setup works flawlessly! The Pi no longer has HDMI output but that's fine for what I'm doing. Thank you again!

Hi. I try to make the looper work with a waveshare screen as well. All works fine until the video is actually played (omxplayer). Could you paste your `boot/config.txt` file ? I try to switch from HDMI to DSI output ? Thx !

gurkanyakar, 2023-09-16:
Hi, when I run the video looper via HDMI everything looks perfect. However, when I try to get an image from the lcd screen connected to the gpio pins, I get a black screen. Is there any way to get back to the console from here? 

When I try to disable dtoverlay=vc4-fkms-v3d I get the same result.

equipment I use 
3.5 inch RPi LCD (C) (connected via GPIO)
Raspberry pi 2 model b

MicaGR83, 2023-12-08:
Hello people,

I would like to apologize if I write something wrong. I am new to all this. Begginer.

Recently I bought a 4 inch

Waveshare DSI-LCD display. I want to use it with a video looper.
I have a problem and the screen is black eveytime i install video looper. Before that, screen is working perfect. I have installed LCD drivers like waveshare says in their wiki and instructions.

Link of the waveshare wiki:
https://www.waveshare.com/wiki/4inch_DSI_LCD_(C)

I am using Video Looper from the below link:
https://videolooper.de/

Is there any way, my raspberry Pi 4 run first all the drivers that I have installed and then run video looper?
How can I fix this problem?

Looking forward for your support.
[image]
[image]