adafruit/pi_video_looper #186: Video Looper not reading changed directory

JarettSutula, 2023-04-08:
A bit of context on what I am trying to do:
I have a project running an API call to a weather API every minute and getting a generic value (sunny, rainy, snowy, etc).
I have .h264 files for each weather condition, and I have two folders - 'playing' which contains the current weather, and then 'videos' which contains all of the other videos. 

Ideally, when my script runs through crontab, it checks the weather and basically swaps the current video out with a different video as the weather changes. At first check, 'playing' only contains sunny.h264 and then a minute later it now only contains rainy.h264 - I can confirm that these swaps do occur in the folders through SSH during the looper and also on the desktop after I shut down the looper. However, even after the videos have been swapped, the looper is still displaying the first video it had when it booted up (in this case, sunny.h264).

I am assuming here that even though the directory has been updated, the video looper is not searching the directory it's been given to look for any new files. Is there anything built in to the looper to re-scour the directory it's been given? (in this case /playing). Any potential command I can run in crontab to force it to refresh the directory?

Thanks again for your time.

tofuSCHNITZEL, 2023-04-08:
Yes, you just run the reload.sh after you placed the new video files in the directory.

JarettSutula, 2023-04-09:
Thanks for your quick response. I tried using reload.sh, but with it quickly flashing a picture of my desktop, it sort of ruined the immersion of my project which is an optical illusion. I think the overlying issue is the way I'm moving files - I was using shutil.move() in python and it appears it may act as a more re-naming action than a straight move action in linux's eyes. I ended up just putting a small sleep between taking the playing video out and the new video being placed in, which seems to force the directory to update twice but doesn't reload the video looper. A small black screen for a second or two is totally fine and I'm happy with the result!

I am new to RPis and this first real project was only really feasible with this video looper - thank you for your work here and for the assistance! 😁

closed 2023-04-09 by JarettSutula