adafruit/pi_video_looper #187: Audio won't work without monitor

jegskjonnerikke, 2023-04-11:
I am having trouble with using the looper just for audio. It won't play without a monitor connected to the hdmi-port. I am currently using it with a DAC (IQaudIO Pi-DAC Pro). Is there any way of solving this issue, so it can loop audio only?

tofuSCHNITZEL, 2023-04-11:
Hi,
I have a suspicion what the problem could, be that it does not run without a HDMI connected... could you provide the log files? (see https://github.com/adafruit/pi_video_looper#troubleshooting)
and can you confirm is also does not output anything on the 3.5mm jack?
cheers

tofuSCHNITZEL, 2023-11-16:
could you maybe change your code like this and see if it plays audio then:
https://github.com/adafruit/pi_video_looper/commit/8a5f43952a6d4ded215e03d8873bed2a704e26ad
this should enable the whole thing to run even if no monitor is connected

closed 2023-11-16 by tofuSCHNITZEL