adafruit/pi_video_looper #188: change audio on mkv video

Crover81, 2023-05-03:
hi, i should have to change audio track of a mkv, it should be possible with omxplayer, but among the enabled keyboard commands there is no this option, it could be a useful option for other people too

Is it possible to do something?
Thanks

Redacted99, 2023-05-11:
Which keys in omxplayer switch streams?  Is it j and k?

Crover81, 2023-05-11:
Exactly,

 j = Previous Audio stream
k = Next Audio stream

Redacted99, 2023-05-12:
I did something similar, I added a pause resume capability using the space key mapped to a omxplayer key command.   Adding j and k support would be similar.  Copy my code and change the bindings and action.  Branch 163 on my fork.  Redacted99/pi_video_looper.

Hope this helps

closed 2023-11-16 by tofuSCHNITZEL