adafruit/pi_video_looper #190: Local Folder not working

Shane-M2, 2023-05-28:
When changing from USB to local directory, there log output shows;

    ModuleNotFoundError: No module named 'Adafruit_Video_Looper./home/pi/video'

It'd be great if the error was more explanatory, or what is involved in using local directory than a USB?  If using a Raspberry Pi Zero, you need to use a OTG cable then a USB.  Would be great to have it all on one little compact setup.

Thanks!

tofuSCHNITZEL, 2023-05-28:
Please provide your ini file.

Shane-M2, 2023-05-28:
The video_looper.ini is stock standard, except for;

# File Reader Location
# Where to find media files.  Can be usb_drive, directory or usb_drive_copymode.
# When using usb_drive any USB stick inserted in to the Pi will be automatically
# mounted and searched for media files (only in the root directory).
# Alternatively the directory option will search only a specified directory on the SD
# card for media files.
# You can change the directory to be used in the [directory] section below.
# The default is usb_drive.
#file_reader = usb_drive
file_reader = /home/pi/video
#file_reader = usb_drive_copymode

Even if I change to a different user and use 'Videos' directory, I get same error.  Files have 'chown 777', so everyone can access the mp4 extension files.

tofuSCHNITZEL, 2023-05-28:
Well this is wrong...
Please read the ini properly it should explain everything.
The location is defined under directory SECTION not at file_reader.

closed 2023-05-28 by tofuSCHNITZEL