adafruit/pi_video_looper #191: GPIO pins for Playback Control etc

jonechkon, 2023-06-14:
Hello,
great stuff this is, thanks!

Just out of curiosity:
Would it be possible to use the gpio pins to control the looper with buttons/potis etc.?
:D

Greetings,
Jochen

shell3000, 2023-06-15:
As long as videolooper supports Keyboards in general, which is great, you can customize HID-Devices to control videolooper. Works fine. Attached the pico keypad from pimoroni.

[image]

jonechkon, 2023-06-17:
Why easy when it can be done complicated...
No, thats true, thanks :)

But after trying it: How would i get the omxplayers keyboard controls working? (especially the speed up/down and seek)

jonechkon, 2023-07-04:
No one? :/

jonechkon, 2023-07-07:
I now have tried to edit the  video_looper.py, adding keybindings.
So far i had been using the precompiled images from https://videolooper.de/ which worked well.
Now after "manual" setup i first need to log into the raspi OS before the looper starts and the keyboard does not respond to neither the original, nor my added keybindings (keyboard_control = true is set in the .ini)
Could that be something i have to setup in the pi OS or would it be because of my wack python editing?

christiansievers, 2023-07-07:
Hmm. Not sure which version of the looper script you need, but be aware that currently the prepackaged image is lagging behind the script on GitHub. If you need the latest version, just update it with git.

jonechkon, 2023-07-09:
Heyyy,
Thanks for your answer :)
I dont know, are there differences in the versions concerning Keyboard commands?

By now i am sure that running the install.sh with additional 
                  if event.key == pygame.K_8:
                    self._player.set_rate(1.5)

... (for example) in the video_looper.py doesnt change anything.
But keyboard input seems to have to be enabled in the OS somehow (how?) anyway, since the standard key settings (p,k,s,ESC...) dont seem to work anyhow after "manual" install.

And on existing systems changing the python files on the sd card does nothing  (for me at least)....

Gruß :D

tofuSCHNITZEL, 2023-07-09:
hey, you dont need to run the full install.sh when changing the code 
`python3 setup.py install --force` in the pi_video_looper dir is sufficient.
check out the https://github.com/adafruit/pi_video_looper/tree/next branch because it contains the code that send keyboard commands directly to omxplayer

jonechkon, 2023-07-10:
Great! Thanks.
I updated the pi and installed the next branch now.
Now the problem is that the videolooper wont start atomatically or after supervisorctl start video_looper although it states "started".

tofuSCHNITZEL, 2023-07-10:
Maybe you did not run install after cloning next and the video_looper.ini is missing settings? What does the log output say?

jonechkon, 2023-07-11:
yep, seems so... how could i...
I also figured out that the videolooper not responding to any keyboard input does have been because of my additions to the video_looper.py and the problem only starts after having pressed one of these newly mapped buttons.

Is there anything obviously wrong with this?
`                if event.key == pygame.K_b:
                    self._print("b was pressed. jumping back...")
                    self._playlist.seek(-1)
                    self._player.stop(3)          
                    #speed&playpos
                if event.key == pygame.K_6:
                    self._player.set_rate(0.062)
                if event.key == pygame.K_1:
                    self._player.set_rate(0.125)
                if event.key == pygame.K_2:
                    self._player.set_rate(0.25)
                if event.key == pygame.K_3:
                    self._player.set_rate(0.5)
                if event.key == pygame.K_4:
                    self._player.set_rate(0.66)
                if event.key == pygame.K_5:
                    self._player.set_rate(0.75)
                if event.key == pygame.K_7:
                    self._player.set_rate(1.3)
                if event.key == pygame.K_8:
                    self._player.set_rate(1.5)
                if event.key == pygame.K_9:
                    self._player.set_rate(2)
                if event.key == pygame.K_0:
                    self._player.set_rate(3)
                if event.key == pygame.K_MINUS:
                    self.player.set_rate(4)
                if event._key == pygame.K_PLUS:
                    self._player.set_rate(6)
                pstmod = 1
                if pygame.key.get_mods() & pygame.KMOD_SHIFT:
                    pstmod = 10
                else:
                    pstmod = 1
                if event.key == pygame.K_KP1:
                    pst = self._player.position()
                    self._player.seek(pst + 1 * pstmod)
                if event.key == pygame.K_KP2:
                    pst = self._player.position()
                    self._player.seek(pst + 1 * pstmod)
....`

tofuSCHNITZEL, 2023-11-16:
GPIO control is now available in https://github.com/adafruit/pi_video_looper/releases/tag/v1.0.14

closed 2023-11-16 by tofuSCHNITZEL