adafruit/pi_video_looper #195: Running a video based on specific time

0xAazaad, 2023-08-04:
I have 2 videos 'a' and 'b'.

I want video a to loop continuously. 
video b to run every even hour (0000 hrs, 0200, hrs, 0400 hrs...) once and then go back to looping a. 

I also want to be able to choose different volumes for both videos. 

Can someone help with achieving this?

Thanks in advance.

shell3000, 2023-08-08:
You can use a script to move your files in and out of you video folder, here /home/pi/video:

#/bin/bash

mv /home/pi/video/b.mp4 /home/pi/video_off
mv /home/pi/video_off/a.mp4 /home/pi/video

# videoduration a.mp4
sleep 30

mv /home/pi/video/a.mp4 /home/pi/video_off
mv /home/pi/video_off/b.mp4 /home/pi/video

Afterwards you modify your crontab to start the script at your given time:

0 0 * * * /bin/bash /home/pi/timer.sh >> /home/pi/timer.log 2>&1
0 2 * * * /bin/bash /home/pi/timer.sh >> /home/pi/timer.log 2>&1
0 4 * * * /bin/bash /home/pi/timer.sh >> /home/pi/timer.log 2>&1
0 6 * * * /bin/bash /home/pi/timer.sh >> /home/pi/timer.log 2>&1

It works very well.

0xAazaad, 2023-08-08:
Thank you. 
What's a good way to autorun this on start?

shell3000, 2023-08-09:
crontab -e

tofuSCHNITZEL, 2023-11-16:
@shell3000 good solution! this is probably the best solution for this besides creating a whole scheduler function/code in video_looper

closed 2023-11-16 by tofuSCHNITZEL