adafruit/pi_video_looper #196: Pi Zero Not Discovering USB - pyudev Issue

td0g, 2023-08-30:
Hi, I successfully installed and am running on a Pi Zero W.  I used the image available at https://videolooper.de/.  

The USB thumb drive was not being discovered by the pi_video_looper script.  No errors were recorded in the logs.  The screen hung on 'Please insert USB....'.  Eventually I was able to resolve the issue by editing a few lines which use the pyudev library.  Simply removing the DEVTYPE='partition' resolved the issue.  The following lines were edited:

https://github.com/adafruit/pi_video_looper/blob/5aad09bb9423915479e941d2b82307d165ce4354/Adafruit_Video_Looper/usb_drive_mounter.py#L37

https://github.com/adafruit/pi_video_looper/blob/5aad09bb9423915479e941d2b82307d165ce4354/Adafruit_Video_Looper/usb_drive_mounter.py#L52

were edited to:

        nodes = [x.device_node for x in self._context.list_devices(subsystem='block')

AND

https://github.com/adafruit/pi_video_looper/blob/5aad09bb9423915479e941d2b82307d165ce4354/Adafruit_Video_Looper/usb_drive_mounter.py#L59

was edited to:

        self._monitor.filter_by('block')

Digging a little deeper, it seems that pyudev was unable to find partitions.  It could still find disks.  I understand this has something to do with the 'udev rules', which I do not understand.  Thank you!

christiansievers, 2023-08-31:
Hi, I've never encountered this error. Respect for finding a fix. 

Can we rule out that there's something unusual about the USB thumb drive? Is it just this one, or the same with others? How were they formatted/partitioned? FAT/exFAT/etc? GUID or MBP partition table?

plugheur33, 2023-09-09:
Hi @td0g thanks for that, I will try this then.
I have the exact same issue. On my side, some USB are working, some are not. They all come from the same supplier. Same format, FAT32. I really don't understand

christiansievers, 2023-09-09:
Could they be broken or fake? (apparently it's easy to fall for fake big-brand USB keys on Amazon)

Still no errors in the logs? https://github.com/adafruit/pi_video_looper#troubleshooting 

Also please run ```sudo dmesg```. 

Can you mount the filesystem? What does ```fdisk -l``` say? 

Then compare the output of those commands to the output you get from the USB keys that do work...

What if you re-format them?

christiansievers, 2023-09-09:
It could also be useful to rule out that it's the DietPi image from videolooper.de that is acting up. What if you install the looper by hand, as described in the Readme here?

plugheur33, 2023-09-09:
I also thought about fake USBs. I bought a Philipps one yesterday, same issue

plugheur33, 2023-09-11:
Hi @christiansievers, tested the above solution and it works on the USBs it wasn't working on, but now no longer works on the USBs it was.

```
Traceback (most recent call last):
 File “/usr/lib/python3.7/runpy.py”, line 193, in _run_module_as_main
  “__main__“, mod_spec)
 File “/usr/lib/python3.7/runpy.py”, line 85, in _run_code
  exec(code, run_globals)
 File “/usr/local/lib/python3.7/dist-packages/Adafruit_Video_Looper-1.0.12-py3.7.egg/Adafruit_Video_Looper/video_looper.py”, line 500, in <module>
 File “/usr/local/lib/python3.7/dist-packages/Adafruit_Video_Looper-1.0.12-py3.7.egg/Adafruit_Video_Looper/video_looper.py”, line 406, in run
 File “/usr/local/lib/python3.7/dist-packages/Adafruit_Video_Looper-1.0.12-py3.7.egg/Adafruit_Video_Looper/video_looper.py”, line 209, in _build_playlist
 File “/usr/local/lib/python3.7/dist-packages/Adafruit_Video_Looper-1.0.12-py3.7.egg/Adafruit_Video_Looper/video_looper.py”, line 218, in _build_playlist_from_all_files
 File “/usr/local/lib/python3.7/dist-packages/Adafruit_Video_Looper-1.0.12-py3.7.egg/Adafruit_Video_Looper/usb_drive.py”, line 30, in search_paths
 File “/usr/local/lib/python3.7/dist-packages/Adafruit_Video_Looper-1.0.12-py3.7.egg/Adafruit_Video_Looper/usb_drive_mounter.py”, line 47, in mount_all
 File “/usr/lib/python3.7/subprocess.py”, line 347, in check_call
  raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command ’[’mount’, ‘-r’, ‘/dev/sda’, ‘/mnt/usbdrive0’]' returned non-zero exit status 32.
```

That is the error we get on the USB it was working on before removing `DEVTYPE='partition'`

Any views on the above? Or how we can have an image where if the option with DEVTYPE does work, it uses it without?

tofuSCHNITZEL, 2023-11-16:
Will close this stale issue

closed 2023-11-16 by tofuSCHNITZEL