adafruit/pi_video_looper #197: Video Looper with 480x1920 monitor issue

designerobot, 2023-09-26:
Hello,

I downloaded the latest version of the Raspberry Pi Video Looper from https://videolooper.de and installed it on a Raspberry Pi 2B. It works perfectly with a standard Full HD monitor (1920x1080).

I tried to get it to work on a 480x1920 monitor named P88 but it doesn't work whatsoever.

With the standard Raspberry installation, the P88 monitor works perfectly. It does not work with DietPi on which Video Looper is based.

The monitor manufacturer's manual says that to work on a Raspberry Pi I need to add these lines to the end of the config.txt file:

```
max_usb_current=1
hdmi_group=2
hdmi_mode=87
hdmi_timings=400 0 100 10 140 1280 10 20 20 2 0 0 0 60 0 43000000 3
```

I added this code at the end of both config.txt and video_looper.ini but nothing changes. 

I also edit the line in video_looper.ini that starts with extra_args and add `--aspect-mode fill` (as read on some forums) but nothing changes again.

Do you have any idea how to make this work?

tofuSCHNITZEL, 2023-11-17:
maybe @christiansievers has an idea why this would not work with Dietpi?
But I would suggest that you just go with the raspbian lite install

designerobot, 2023-11-18:
Thanks @tofuSCHNITZEL, I can install Raspbian Lite... but then what should I do to create a video looper that works automatically as just as I plug in the power? Do you have a complete tutorial to suggest maybe?

tofuSCHNITZEL, 2023-11-18:
The readme on this repo explains how to install...
https://github.com/adafruit/pi_video_looper/#how-to-install

Or here
https://learn.adafruit.com/raspberry-pi-video-looper/installation

Please use raspian lite image linked in the readme.