adafruit/pi_video_looper #198: Implementing GPIO controls

spikereloaded, 2023-10-05:
I'm wanting to use GPIO buttons to control my always on cartoon player and struggling to get the events to register and work with the program. 

I'm using similar to this code but I'm unsure where each part should go or even if an interrupt event can be used to trigger events in the looper script. I've been putting it at different places with the video_looper.py script with no success.  

I'm not at my computer currently so unable to give my exact code but i basically want to be able to skip, skip backwards and potentially adjust system volume as well, using maybe 4 different GPIO buttons. 
```
import RPi.GPIO as GPIO # Import Raspberry Pi GPIO library

def button_callback(channel):
     print("Button was pushed!")
     self.quit()

GPIO.setwarnings(False) # Ignore warning for now
GPIO.setmode(GPIO.BOARD) # Use physical pin numbering
GPIO.setup(10, GPIO.IN, pull_up_down=GPIO.PUD_DOWN) # Set pin 10 to be an input pin and set initial value to be pulled low (off)

GPIO.add_event_detect(10,GPIO.RISING,callback=button_callback) # Setup event on pin 10 rising edge
```

tofuSCHNITZEL, 2023-11-16:
Hi,
check out
https://github.com/adafruit/pi_video_looper/tree/buttons
its untested at the moment but I will try to confirm function in the next couple of days.

tofuSCHNITZEL, 2023-11-16:
I have tested and released version v1.0.14 that includes GPIO control - I hope it will fit your usecase!

closed 2023-11-16 by tofuSCHNITZEL