adafruit/pi_video_looper #203: Using 2 monitors on Pi4

Chrisgraue, 2023-11-18:
This is a feature request, but would be really cool:

The Pi4 has two HDMI outputs. I'd love to be able to send either a split video (a single 3840x1080 video) or two videos (each 1920x1080) to different monitors at the same time.

I found one person who worked out an implementation of this for OMXplayer, but only wrote it for a single video. Perhaps if the looper has something that could be used to dynamically change which video is targeted in this? It might be a starting point.

DISPLAY=:0 xterm -e omxplayer --display=2 /home/pi/Videos/video2.mp4 & omxplayer --display=7 /home/pi/Videos/video2.mp4

tofuSCHNITZEL, 2023-11-19:
Since omxplayer is EOL and will not be installable in the future I would not get my hopes up (and cvlc is not an easy replacement)
But the trick is to just run the video looper twice with different inis and in the inis use the extra args to have it display the video on different screens.
see also here:
https://github.com/adafruit/pi_video_looper/issues/100

tofuSCHNITZEL, 2023-11-19:
you can check out the readme here: https://github.com/adafruit/pi_video_looper/tree/next#raspberry-pi-45-multi-screen-output
but note this is totally untested yet!

Chrisgraue, 2023-11-19:
Interesting! I wonder if I could point each instance to a different folder, basically an A1 and B1 playing at the same time, A2 and B2, etc etc. If I'm lucky, hitting the controls to skip, pause, back, etc will control both of them.

tofuSCHNITZEL, 2023-11-19:
yes different folders is no problem (just edit the ini of one of the instances)
if KB control would control both instances I don't dare predict ;) needs to be tested ;)