adafruit/pi_video_looper #204: Chapters for gapless playback

Chrisgraue, 2023-11-19:
I might have an idea to make a rudimentary gapless playback: chapters.

OMXplayer supports chapters, and skipping chapters with the O and I keys. If you make one really long video file with set chapter points, it should all conceivably be loaded and may reduce/eliminate the gap from skipping files.

I've tried testing it, but it seems the looper currently does not recognize O and I key inputs. I'm wondering if it patching it would work by the same method the pause/play space command was added.

tofuSCHNITZEL, 2023-11-19:
Hi,
currently omxplayer already supports gapless loops of a single file.
gaps between files are a side effect of how the script operates and the gap can't be eliminated with the current code.
what would you like to achieve with the chapter select exactly?
you can send these key commands to the omxplayer process exactly like the pause command is currently sent - so its easy to modify the code to do what you want.
besides omxplayer is EoL - it can't be installed on newer images and the move to cvlc is evermore pressing.

Chrisgraue, 2023-11-19:
Yeah, cvlc is definitely a bigger priority. I just thought this would be an easy thing since it's just adding a button press currently supported by the software. VLC also supports next/prev chapter, but with a different key.

I use the player to do a live concert, with a pi playing my tracks/video. When I click the next button, there's a moment of black as it cues up the next track, which is a bit distracting and takes the audience out of the world I create (in which I'm battling a computer over which I have no control).

My thought is that by using chapters within a single video file, it might seamlessly advance to the next timecode of the video that has a chapter marker on it (easy to add in Premiere or any video editor). And that way, since it's technically one file instead of several, it should be able to gapless loop on itself.

tofuSCHNITZEL, 2023-11-19:
Thanks for clarifying your use case! 
Check out the current commit to https://github.com/adafruit/pi_video_looper/tree/next
I have implemented the sending of o/i to omxplayer - if you have time please test it ;)

Chrisgraue, 2023-12-06:
I tried following the same install commands, but instead pointing it to https://github.com/adafruit/pi_video_looper/tree/next

and it said the directory could not be found. Is there something I'm missing to install this version?

tofuSCHNITZEL, 2023-12-06:
This is not the adress you use for cloning different branches...
You use the regular url and add a `-b branchname` to the git clone command.

`git clone --single-branch --branch next https://github.com/adafruit/pi_video_looper/`

Chrisgraue, 2023-12-08:
Installed and tested! Does not appear to work in passing along chapter skips. OR possibly OMXplayer does not understand the chapter format used in h264 and cannot see that there are chapters to skip. It was worth a shot! Thank you

tofuSCHNITZEL, 2023-12-08:
I didn't have a file ready with chapters so I could not test it...

tofuSCHNITZEL, 2023-12-08:
Have you maybe tried creating an mkv file with chapters?
https://www.reddit.com/r/VideoEditing/comments/p3fc61/how_do_you_add_chapters_to_a_video_file/
Maybe this works better with omxplayer

tofuSCHNITZEL, 2024-02-13:
okay short update...
I made a stupid error in the code so it could not work. And now I also tested it with a file I added chapters to. (with this great tool: https://github.com/stevie910/drax/releases)

closed 2024-02-13 by tofuSCHNITZEL