adafruit/pi_video_looper #207: step-by-step guide for updating pi_video_looper

SamVimes78, 2023-12-20:
Can you please add a guide to the readme (and maybe also on the adafruit homepage) on how to check what version of pi_video_looper is installed and how to upgrade to the latest version?

tofuSCHNITZEL, 2023-12-20:
the upgrade path is the same for an install - you just need to remove the cloned pi_video_looper folder first

SamVimes78, 2023-12-20:
This information is certainly very useful for someone that knows how git works.

Let's just assume i don't. And maybe one or two others that installed the video looper by following the instructions on the adafruit homepage also don't...

tofuSCHNITZEL, 2023-12-20:
okay point taken - but these people should maybe try to understand the bare minimum of what they are doing, even if they are just following a recipie... 
its like wanting to drive a car without knowing how to fuel it - eventually this will become a problem

tofuSCHNITZEL, 2023-12-20:
added update steps to the readme - if you think it should include more / other stuff please let me know

closed 2023-12-20 by tofuSCHNITZEL

SamVimes78, 2023-12-20:
Thanks for that! But I think it has to be 
`sudo rm -rf pi_video_looper`

It's a bit inconvenient to lose the video_looper.ini. But I know THAT much about the usage to set up everything again. And if a new version adds new functions to the .ini it may be good to reset it.

Maybe you could add
`sudo cp /boot/video_looper.ini /boot/video_looper.ini_backup`
to the guide
just to give people a chance to look up their old settings.

SamVimes78, 2023-12-20:
FYI i added an answer to the closed issue.

tofuSCHNITZEL, 2023-12-20:
> Thanks for that! But I think it has to be `sudo rm -rf pi_video_looper`
>
> It's a bit inconvenient to lose the video_looper.ini. But I know THAT much about the usage to set up everything again. And if [...]

done. thanks for the info!