adafruit/pi_video_looper #209: GPIO input for player and device command

anoril1979, 2024-01-11:
It is possible to control the RaspPi with keyboard, like 'p' to shutdown, and have GPIO interrupts to change video played.

Is it possible to add RaspPi commands (shutdown) to GPIO interrupts ?

Regards,

tofuSCHNITZEL, 2024-01-11:
yes did you even read the readme?
https://github.com/adafruit/pi_video_looper#control

you can edit the code to make the gpio also run the shutdown action.

anoril1979, 2024-01-11:
Hi!
Great!

Sorry, I'm new to RaspberryPi programming, and yes: I read the readme which I found not really clear on this point (it literally says you have actions on keys, and GPIO Can be mapped to filenames, file indexes or relative jumps :) )

Many thanks for your time replying my request and for this great piece of work!

And I wish you the best for the New Year!
Regards

closed 2024-01-11 by anoril1979