adafruit/pi_video_looper #210: Timer Control

SamVimes78, 2024-01-19:
Intro (feel free to skip):
We're running multiple Pi Zeros looping happily through some videos in our store. I set them up (cron) to stop video_looper 5 minutes after closing each day and sending the attached tv into standby another minute later. 5 minutes before we open the RasPis will reboot and wake up the TVs again.

Request:
I'd like video_looper to switch to a certain video at a certain time (without needing to stop video_looper and showing command prompt). That way i'd be able to show a message that our store will be closing in a few minutes.

tofuSCHNITZEL, 2024-01-20:
Hey,

you can use cron to just swap the files in the folder for the video you want to have played at the certain time - this would cause videolooper to start the playback of the file without having to reload the looper (in directory mode the directory is watched and playback changes if files change)

Also I share with you now the changes I make to all the PIs I use with the videolooper to hide the bootscreen and the terminal:

```
#HIDE splashscreen
sudo sh -c "echo 'disable_splash=1' >> /boot/config.txt"

#DISABLE WARNINGS:
sudo sh -c "echo 'avoid_warnings=1' >> /boot/config.txt"

#quiet boot
sudo sed -i -e 's/rootwait/rootwait quiet splash loglevel=0 consoleblank=0 logo.nologo vt.global_cursor_default=0 plymouth.ignore-serial-consoles /' /boot/cmdline.txt

#Tell dmesg to be quiet
sudo sed -i  '/exit 0/i\#Suppress Kernel Messages\ndmesg --console-off\n' /etc/rc.local

#hide motd
touch ~/.hushlogin

#SILENT BOOT AND SPLASH
sudo systemctl disable getty@tty1
```

I also install a "splashscreenservice" which just displayes a custom fullscreen PNG instead of a black screen until the videolooper starts. if you want I can share this also with you

christiansievers, 2024-01-21:
Hi @tofuSCHNITZEL I've finally found time to update my image file for videolooper.de again, and your script looks like it would be good to include as a manual option. If there's anything else you want to share, let's have it! :)  
(sorry for hijacking this ticket)

tofuSCHNITZEL, 2024-01-21:
> Hi @tofuSCHNITZEL I've finally found time to update my image file for videolooper.de again, and your script looks like it would be good to include as a manual option. If there's anything else you want to share, let's have it! :) (sorry for hijacking this ticket)

here is what I do for the "boot up screen":

```
sudo cp ./assets/splashscreen.service /etc/systemd/system/splashscreen.service
sudo cp ./assets/loader.png /home/pi/loader.png

sudo systemctl enable splashscreen
```

splashscreen.service looks like this:

[Unit]
Description=Splashscreen
DefaultDependencies=no
After=local-fs.target

```
[Service]
ExecStart=/usr/bin/fbi -d /dev/fb0 --noverbose -a /home/pi/loader.png
StandardInput=tty
StandardOutput=tty

[Install]
WantedBy=sysinit.target
```

package "fbi" needs to be installed