adafruit/pi_video_looper #211: Video Looper doesn boot (using analog video adapters vga666/rgb-pi) but running reload.sh fixes it.

fixongbg, 2024-01-29:
Hello! 

I've been troubleshooting for two weeks now so I need some help. I'm trying to play videos on my CRT through my vga666 adapter (made for CRTs using SCART) and I also tried this RGB-Pi adapter. During console boot I get a picture with the adapter, I can see the console starting so the adapters work but when Video Looper is suppose to start I only get a black image (this is where the "insert media text" is shown). 
Although, if I run the script **reload.sh** , video looper starts and analog video works though my adapters.
I'm confused why this is happening when using analog output video through the GPIOs (which the adapters requires). When using HDMI it works perfect!

I'm running the latest image from videolooper.de on a Rpi 3B. 

Here's my config.txt file:

dtparam=audio=on
dtoverlay=pwm-2chan,pin=18,func=2,pin2=19,func2=2
dtoverlay=rgb-pi
enable_dpi_lcd=1
display_default_lcd=1
dpi_output_format=6
dpi_group=2
dpi_mode=87
hdmi_timings=320 1 16 30 34 240 1 2 3 22 0 0 0 60 0 6400000 1

Parameters I've changed in video_looper.in are:
sound = local 
password =

christiansievers, 2024-01-29:
Hi, I just updated the image on videolooper.de a few days ago. It's now on its version 2.7, using the latest pi_video_looper [v1.0.15]. Would you try that?

fixongbg, 2024-01-29:
I using the v2.7 and download it yesterday actually!