adafruit/pi_video_looper #212: Raspberry PI5 not supported

nonuser666, 2024-02-03:
I have tried the latest v.2.7 – January 2024, does not  work on PI5

I tried the recommended OS, not working.

Lastly tried the modify the config.txt---> os_check=0. 

Resulted to black screen.

Feel free to contact me for testing the .img.

christiansievers, 2024-02-04:
Hey, thanks for the feedback. Just to be sure - do you mean both the [ready-made image ](https://videolooper.de/) and the default Raspbian OS legacy image (as recommended in the [readme](https://github.com/adafruit/pi_video_looper?tab=readme-ov-file#pi_video_looper) here) don't work?

nonuser666, 2024-02-04:
Hi. Exactly, both not working.

I think because the Pi5 does not support the legacy OS.

tofuSCHNITZEL, 2024-02-04:
Install works fine? Have you tried to enable fake kms via rpi config?

nonuser666, 2024-02-04:
Install worked fine. I only disabled os_check

tofuSCHNITZEL, 2024-02-04:
okay please try fake kms and report back

tofuSCHNITZEL, 2024-02-04:
> I need few days. Can send me a how to for the fake kms i am new to Pi. A video or an article would be great.
> […](#)

see above https://fydeos.io/docs/knowledge-base/recipes/enable-fake-kms-for-raspberry-Pi/)

AaronMcEntire, 2024-12-09:
Any follow up on this?  I've ordered some 5's and was hoping to use Looper...

tofuSCHNITZEL, 2024-12-09:
nope. but you could try to install the image mentioned in the readme update the drivers on the sd card following this instructions:
with luck the old software will boot on the new pi5
```
Updating only the firmware
Sometimes, going through the standard upgrade procedure may not be possible. For example, you may have a
customised distribution with no update facilities, or that cannot be upgraded without causing further issues. In these
circumstances, and possibly others, you will need to update the firmware files in the distribution manually.
You can download the firmware files from the following location:
https://github.com/raspberrypi/firmware/archive/refs/heads/stable.zip
This zip file contains a number of items, but the ones we are interested in at this stage are in the boot folder. The firmware
files have names of the form start*.elf, and their associated support files are fixup*.dat.
The basic principle is to copy the required start and fixup files from this zip file to replace the same named files in the
destination OS image. The exact process will depend on how the OS has been set up, but this is an example of how it
would be done for a Raspberry Pi OS image:
1. Extract or open the zip file so you can access the required files.
2. Open up the boot folder in the destination OS image (this could be on an SD card, or a disk-based copy).
3. Determine which start.elf and fixup.dat files are present in the destination OS image.
4. Copy those files from the zip archive to the destination image.
The image should now be ready for use on the latest Raspberry Pi Ltd hardware.

```

marbles-pi, 2025-05-15:
@tofuSCHNITZEL Is it just the complete image that doesn't work on the Pi5, or does it also not work when you install the script/application on a pi5 yourself?
I have used pi_video_looper a lot - always on the default PI OS for that type and then adding the script myself.... 
Thanks!

tofuSCHNITZEL, 2025-05-15:
I have never tried the complete image but I imagine you could make the self install work by using the old image (linked in the readme) and add the new firmware files (see post above)

marbles-pi, 2025-05-15:
> I have never tried the complete image but I imagine you could make the self install work by using the old image (linked in the readme) and add the new firmware files (see post above)

I'm having trouble understanding this answer... 
Can't I just:
- buy Pi 5
- install Raspberry Pi OS 12 (bookworm)
- git clone pi_video_looper.git
- install.sh
?
This is what always worked for me?

tofuSCHNITZEL, 2025-05-16:
I would recommend you get a 3b+ or if you want USB-C and microHDMI get the rpi4 but there is no advantage to having a pi5.
you cannot install Pi OS 12 because omxplayer (the main component of the looper) does not exist there anymore.
there is a link to the last pi os image that includes omxplayer but this will not work on a pi5 if you don't do this (https://github.com/adafruit/pi_video_looper/issues/212#issuecomment-2529610560) after flashing it to the sdcard.