adafruit/pi_video_looper #213: GPIO Safe shutdown option

ArgonautLeader, 2024-02-10:
**Add the ability to use a momentary button connected via GPIO17 to safely shutdown the Pi.**

I'm planning to use the Video Looper to make cheap video displays to be deployed at a small Heritage Centre (aka museum) using Pi Zeros mounted onto the rear of 2nd hand monitors (the Centre is a registered charity so money is scarce).  

However as the Stewards who host the Centre when it is open are not technically minded I need a simple way for them to shut down the displays at the end of the day without the risk of corrupting the OS on the video card.  

So I was thinking by adding a momentary button that triggers a safe shut down before they turn the monitor off at the wall socket would be the ideal solution - their "shutting the Centre written instructions would be updated to tell them to wait a period of time (30 sec ?) or until the power on LED extinguishes until the Zero has safely shut down.  

I can mount the button (and possible a power on LED) into the monitors case as I'm already planning on cracking open the case to obtain a 240V AC feed for a small 5V PSU to power the Zero so adding a small momentary switch, power led and associated wiring will not be a challenge (I'm more an electronics/hardware guy)

I have found this article on the Sparkfun web site which covers exactly the above but I'm not that clued up with Python, so if somebody could add this to the Image I would be really grateful.

https://learn.sparkfun.com/tutorials/raspberry-pi-safe-reboot-and-shutdown-button/all#:~:text=Press%20and%20hold%20GPIO17%20button,your%20finger%20off%20the%20button.

Thank you

Redacted99, 2024-02-10:
At our historical society we just tell the docents to quit the looper and we have the looper shutdown the system on exit

I think this is standard functionality

ArgonautLeader, 2024-02-10:
> At our historical society we just tell the docents to quit the looper and we have the looper shutdown the system on exit
> 
> I think this is standard functionality

I'm assuming that is via the keyboard shortcut?

Redacted99, 2024-02-10:
Yes, using key
“p" - Power off - stop playback and shutdown RPi

ArgonautLeader, 2024-02-10:
> Yes, using key “p" - Power off - stop playback and shutdown RPi

Yes but having a keyboard connected to every display isn't exactly cost effective or user friendly regarding non computer literate Stewards.  Much easier to say " just press the red button". Also a display with a keyboard connected isn't very aesthetically pleasing in a museum environment IMHO.

tofuSCHNITZEL, 2024-02-10:
its very straight forward to modify the code to have the shutdown mapped to a gpio port.
but I will see if I have some time tomorrow and I will make it mapable via the ini file.

ArgonautLeader, 2024-02-10:
> its very straight forward to modify the code to have the shutdown mapped to a gpio port. but I will see if I have some time tomorrow and I will make it mapable via the ini file.

Thank you it's very appreciated if you do manage to find the time 👍

tofuSCHNITZEL, 2024-02-11:
I just pushed an update to the https://github.com/adafruit/pi_video_looper/tree/next branch.
But beware - its currently untested... I would appreciate feedback if possible.
you can follow the readme but use this clone command instead:
`git clone -b next https://github.com/adafruit/pi_video_looper`

ArgonautLeader, 2024-02-11:
> I just pushed an update to the https://github.com/adafruit/pi_video_looper/tree/next branch. But beware - its currently untested... I would appreciate feedback if possible. you can follow the readme but use this clone command instead: `git clone -b next https://github.com/adafruit/pi_video_looper`

That is awesome, once I have it set up (still waiting on a mini HDMI cable to arrive) I will give it a go and feed back.  I really appreciate it, Thank you.

tofuSCHNITZEL, 2024-02-12:
okay I have found an issue an updated the next branch, but I also tested it now on my local setup - looks good so far.

tofuSCHNITZEL, 2024-02-13:
this is now part of version [1.0.17](https://github.com/adafruit/pi_video_looper/releases/tag/v1.0.17)

closed 2024-02-13 by tofuSCHNITZEL