adafruit/pi_video_looper #216: GPIO Folder Control

Blake0206, 2024-02-21:
I was writing to suggest a feature that would allow for a new folder of videos to be played in random order when GPIO is pressed. If there is a way to modify the code to make that work then I would appreciate it! Maybe something to do with a new playlist being selected on GPIO press.

cstoney, 2024-05-13:
I'm looking for something similar. I was thinking playlist, but folders would be easier for organization. For example:

GPIO 15 : "assets/folder1" (begins playing videos from "folder1")

But to add to this, I'd like to be able to jump to the next folder (or playlist) with another input. Such as:

GPIO 16 : "+1 folder" or "+1 playlist" (switches to the next source alphabetically - folder1 > folder2, etc)

pelicanmedia, 2024-09-24:
+1 for this
ie something like the below:
`gpio_pin_map = "11" : "playlist_1.m3u", "13": "playlist_2.m3u", "16": "playlist_on_start.m3u"`