adafruit/pi_video_looper #217: GPIO not working

Funkelfetisch, 2024-03-21:
my line:
```
gpio_pin_map = "3" : "gpio.mp4"
```

is apparently read, because the internal pull-resistor switches from pull-down to pull-up, which I can see with raspi-gpio.

I can also read the gpio fine with raspi-gpio get -- yet nothing happens in the video looper.

so I assume there's a bug in the code, right?

tofuSCHNITZEL, 2024-03-21:
Could you enable console output in the ini and check what is written to the logs? (see bottom of readme on how)
There we should see if the gpio is actually triggered. Are you sure that the file is actually in the correct location? (drive or directory?)

BenjaminPoilve, 2024-10-17:
I seem to have a similar problem. 
I'm using the image provided by Christian Sievers on [this website](https://videolooper.de/). 
It's based on the 1.0.17 version on a 2022-01-28-raspios-buster-armhf-lite. I'm using a Raspberry PI B+. 
Using raspi-gpio, I can see that my switch indeed pulls down the targeted pin (after appropriately setting the pull up). 
Activating the log, I can see that the pin is indeed configured as I wish but... it never triggers. 
I tried re-running the install.sh script, just in case, but it does not change anything. 
I'm a bit lost on how to investigate further. If you have any pointer, it might useful !