adafruit/pi_video_looper #219: Video plays with omxplayer directly, but script shows black screen

githubarooski, 2024-05-03:
If I call omxplayer directly with my custom resolution, it works fine. If I use your script to call omxplayer with the same arguments, it just shows a black screen and it seems to keep crashing based on looking at `ps aux` 

If I kill your script and then run it manually, I see this:

```
root@latent:/home/pi/pi_video_looper# python3 -u -m Adafruit_Video_Looper.video_looper
pygame 1.9.4.post1
Hello from the pygame community. https://www.pygame.org/contribute.html
Starting Adafruit Video Looper.
```

You can see that the script is using my custom args successfully from your config file:
```

root@latent:/home/pi/pi_video_looper# ps axu | grep omx
root      4084  0.0  0.0   7672  2952 ?        S    17:35   0:00 /bin/bash /usr/bin/omxplayer -o both --win "0 0 1600 1200" -g --loop /home/pi/video/video.mp4
root      4095  0.0  0.3 108840 14972 ?        RL   17:35   0:00 /usr/bin/omxplayer.bin -o both --win "0 0 1600 1200" -g --loop /home/pi/video/video.mp4
root      4097  0.0  0.0   7344   508 pts/0    S+   17:35   0:00 grep omx
```

But nothing appears on my screen and I can't find the omxplayer.log file on the filesystem anywhere, so I don't think it's creating the log file, even though I am passing it `-g` in the custom arguments config. 

When I run the command manually, the video plays just fine

```
root@latent:/home/pi/pi_video_looper# /bin/bash /usr/bin/omxplayer -o both --win "0 0 1600 1200" -g --loop /home/pi/video/video.mp4
Video codec omx-h264 width 616 height 816 profile 100 fps 8.000000
Subtitle count: 0, state: off, index: 1, delay: 0
V:PortSettingsChanged: 616x816@8.00 interlace:0 deinterlace:0 anaglyph:0 par:1.00 display:0 layer:0 alpha:255 aspectMode:0
```

When I run my omxplayer command manually and create a log file, I see this in the log file

```
# cat omxplayer.log 
﻿09:32:32 T:1714500514   DEBUG: DllBcm: Using omx system library
09:32:32 T:1714501705   DEBUG: DllOMX: Using omx system library
09:32:32 T:1714502345   DEBUG: DllAvFormat: Using libavformat system library
09:32:32 T:1714503696   DEBUG: DBus connection succeeded
09:32:32 T:1714504452   DEBUG: Keyboard: DBus connection succeeded
09:32:32 T:1714504614   DEBUG: OMXThread::Create - Thread with id -1388339520 started
09:32:32 T:1714504716   DEBUG: DllAvUtilBase: Using libavutil system library
09:32:32 T:1714504739   DEBUG: DllAvCodec: Using libavcodec system library
09:32:32 T:1714504756   DEBUG: DllAvFormat: Using libavformat system library
09:32:32 T:1714546422   DEBUG: COMXCoreComponent::Initialize OMX.broadcom.clock input port 80 output port 81 m_handle 0x1385990
09:32:32 T:1714546673   DEBUG: OMXClock::OMXStop
09:32:32 T:1714546782   DEBUG: OMXClock::OMXSetSpeed(0.00) pause_resume:1
09:32:32 T:1714546977   DEBUG: DllAvUtilBase: Using libavutil system library
09:32:32 T:1714547000   DEBUG: DllAvCodec: Using libavcodec system library
09:32:32 T:1714547018   DEBUG: DllAvFormat: Using libavformat system library
09:32:32 T:1714548272   DEBUG: COMXCoreComponent::Initialize OMX.broadcom.video_decode input port 130 output port 131 m_handle 0x13858a0
09:32:32 T:1714549126   DEBUG: COMXCoreComponent::AllocInputBuffers component(OMX.broadcom.video_decode) - port(130), nBufferCountMin(1), nBufferCountActual(60), nBufferSize(81920), nBufferAlignmen(16)
09:32:32 T:1714558318   DEBUG: COMXVideo::Open - decoder_component(0x0x13858a0), input_port(0x82), output_port(0x83) deinterlace 1 hdmiclocksync 0
09:32:32 T:1714558538   DEBUG: OMXThread::Create - Thread with id -1400901952 started
09:32:32 T:1714558600  NOTICE: OMXClock using video as reference
09:32:32 T:1714558828   DEBUG: OMXClock::OMXReset audio / video : 0 / 1 wait mask 0->2 state : 2->1
09:32:32 T:1714559131   DEBUG: Popped message member: NameAcquired interface: org.freedesktop.DBus type: 4 path: /org/freedesktop/DBus
09:32:32 T:1714559170 WARNING: Unhandled dbus message, member: NameAcquired interface: org.freedesktop.DBus type: 4 path: /org/freedesktop/DBus
09:32:32 T:1714559284   DEBUG: Normal M:0 (A:0 V:-4503599627370496) P:1 A:0.00 V:0.00/T:0.20 (0,0,1,0) A:0% V:0% (0.00,0.00)
09:32:32 T:1714559636    INFO: CDVDPlayerVideo::Decode dts:-250000 pts:0 cur:0, size:94041
09:32:32 T:1714559749   DEBUG: OMXVideo::Decode VDec : setStartTime 0.000000
09:32:32 T:1714560669    INFO: CDVDPlayerVideo::Decode dts:-125000 pts:500000 cur:500000, size:104944
09:32:32 T:1714561643    INFO: CDVDPlayerVideo::Decode dts:0 pts:250000 cur:250000, size:98915
09:32:32 T:1714562610    INFO: CDVDPlayerVideo::Decode dts:125000 pts:125000 cur:125000, size:30321
09:32:32 T:1714563186    INFO: CDVDPlayerVideo::Decode dts:250000 pts:375000 cur:375000, size:28025
09:32:32 T:1714563688    INFO: CDVDPlayerVideo::Decode dts:375000 pts:1000000 cur:1000000, size:106607
09:32:32 T:1714564826    INFO: CDVDPlayerVideo::Decode dts:500000 pts:750000 cur:750000, size:103248
09:32:32 T:1714566167    INFO: CDVDPlayerVideo::Decode dts:625000 pts:625000 cur:625000, size:23891
09:32:32 T:1714567098    INFO: CDVDPlayerVideo::Decode dts:750000 pts:875000 cur:875000, size:27867
09:32:32 T:1714567818    INFO: CDVDPlayerVideo::Decode dts:875000 pts:1500000 cur:1500000, size:104816
[... 293 lines cut ...]
09:32:36 T:1718479513   DEBUG: Normal M:3699061 (A:0 V:13000000) P:0 A:-3.70 V:9.30/T:0.20 (0,0,1,1) A:0% V:99% (0.00,0.00)
09:32:36 T:1718499795   DEBUG: Normal M:3719553 (A:0 V:13000000) P:0 A:-3.72 V:9.28/T:0.20 (0,0,1,1) A:0% V:99% (0.00,0.00)
09:32:36 T:1718520075   DEBUG: Normal M:3739833 (A:0 V:13000000) P:0 A:-3.74 V:9.26/T:0.20 (0,0,1,1) A:0% V:99% (0.00,0.00)
09:32:36 T:1718540357   DEBUG: Normal M:3760114 (A:0 V:13000000) P:0 A:-3.76 V:9.24/T:0.20 (0,0,1,1) A:0% V:99% (0.00,0.00)
09:32:36 T:1718560638   DEBUG: Normal M:3780396 (A:0 V:13000000) P:0 A:-3.78 V:9.22/T:0.20 (0,0,1,1) A:0% V:99% (0.00,0.00)
09:32:36 T:1718581126   DEBUG: Normal M:3800677 (A:0 V:13000000) P:0 A:-3.80 V:9.20/T:0.20 (0,0,1,1) A:0% V:99% (0.00,0.00)
09:32:36 T:1718601410   DEBUG: Normal M:3821169 (A:0 V:13000000) P:0 A:-3.82 V:9.18/T:0.20 (0,0,1,1) A:0% V:99% (0.00,0.00)
09:32:36 T:1718621691   DEBUG: Normal M:3841450 (A:0 V:13000000) P:0 A:-3.84 V:9.16/T:0.20 (0,0,1,1) A:0% V:99% (0.00,0.00)
09:32:36 T:1718641972   DEBUG: Normal M:3861730 (A:0 V:13000000) P:0 A:-3.86 V:9.14/T:0.20 (0,0,1,1) A:0% V:99% (0.00,0.00)
09:32:36 T:1718659140    INFO: CDVDPlayerVideo::Decode dts:12875000 pts:13000000 cur:13000000, size:43223
09:32:36 T:1718659853    INFO: CDVDPlayerVideo::Decode dts:13000000 pts:13625000 cur:13625000, size:74990
09:32:36 T:1718662261   DEBUG: Normal M:3882019 (A:0 V:13375000) P:0 A:-3.88 V:9.49/T:0.20 (0,0,1,1) A:0% V:98% (0.00,0.00)
09:32:36 T:1718683537   DEBUG: Normal M:3903087 (A:0 V:13375000) P:0 A:-3.90 V:9.47/T:0.20 (0,0,1,1) A:0% V:99% (0.00,0.00)
09:32:36 T:1718703844   DEBUG: Normal M:3923604 (A:0 V:13375000) P:0 A:-3.92 V:9.45/T:0.20 (0,0,1,1) A:0% V:99% (0.00,0.00)
09:32:36 T:1718714119   DEBUG: OMXClock::OMXStop
09:32:36 T:1718730038   DEBUG: OMXThread::Run - Exited thread with  id -1400901952
09:32:36 T:1718730653   DEBUG: OMXThread::StopThread - Thread stopped
09:32:36 T:1718753322   DEBUG: COMXCoreComponent::Deinitialize : OMX.broadcom.video_scheduler handle 0xac900cb0
09:32:36 T:1718766500   DEBUG: COMXCoreComponent::Deinitialize : OMX.broadcom.video_decode handle 0x13858a0
09:32:36 T:1718768603   DEBUG: COMXCoreComponent::Deinitialize : OMX.broadcom.video_render handle 0xac900670
09:32:36 T:1718768973   DEBUG: OMXThread::Run - Exited thread with  id -1388339520
09:32:36 T:1718769284   DEBUG: OMXThread::StopThread - Thread stopped
09:32:36 T:1718772825   DEBUG: COMXCoreComponent::Deinitialize : OMX.broadcom.clock handle 0x1385990

```

githubarooski, 2024-05-03:
for now I'm just hard coding it like this, but any tips on fixing it properly would be appreciated

```
$ cat /etc/supervisor/conf.d/video_looper.conf 
# Supervisord configuration to run video looper at boot and
# ensure it runs continuously.
[program:video_looper]
command=/bin/bash /usr/bin/omxplayer -o both --win "0 0 1600 1200" -g --loop /home/pi/video/video.mp4
#command=python3 -u -m Adafruit_Video_Looper.video_looper
autostart=true
autorestart=unexpected
startsecs=5
```

githubarooski, 2024-05-03:
> Hi have you tried adding `--win "0 0 1600 1200"` to the videolooper.ini? there is a setting called extra_args that is passed to omxplayer and I think this is exactly what you need in this case...

Thanks for your quick response. If you read my report, you can see that I've added the argument to the extra_args config and your script is adding those arguments correctly. 

Can you help me figure out how to get omxplayer to generate a log file when it is called from your script? I add the `-g` to the extra_args, but I can't find the log file anywhere.

tofuSCHNITZEL, 2024-05-03:
sorry for missing that..
please check the output of the videolooper log file like described here:
https://github.com/adafruit/pi_video_looper?tab=readme-ov-file#troubleshooting
maybe we can also see the omxplayer output there

tofuSCHNITZEL, 2024-05-03:
hm, I read your post now properly - not sure why you would get a black screen as far as I can see pivideolooper runs omxplayer exactly like you so there is no reason why it would not display an image... not that it should matter (because it works with directly calling omx) what OS are you using? is the screen connected to a normal hdmi output?

githubarooski, 2024-05-03:
Using HDMI0 of a raspberry pi 4 connected to the HDMI input of a very niche e-ink monitor from philips. 

Here are the only active entries in my config.txt which I pulled from a similar monitor's info via WaveShare.  
```
max_usb_current=1
hdmi_group=2
hdmi_force_hotplug=1
hdmi_mode=87
hdmi_timings=1600 1 64 192 144 1200 1 4 5 41 0 0 0 60 0 150000000 1
```

nothing is written to video_looper-stderr.. 

supervisord.log
```
2024-05-03 19:55:50,631 INFO waiting for video_looper to die
2024-05-03 19:55:50,664 WARN received SIGTERM indicating exit request
2024-05-03 19:55:51,863 INFO stopped: video_looper (exit status 0)
2024-05-03 19:55:52,321 CRIT Supervisor is running as root.  Privileges were not dropped because no user is specified in the config file.  If you intend to run as root, you can set user=root in the config file to avoid this message.
2024-05-03 19:55:52,321 INFO Included extra file "/etc/supervisor/conf.d/video_looper.conf" during parsing
2024-05-03 19:55:52,343 INFO RPC interface 'supervisor' initialized
2024-05-03 19:55:52,344 CRIT Server 'unix_http_server' running without any HTTP authentication checking
2024-05-03 19:55:52,344 INFO supervisord started with pid 9882
2024-05-03 19:55:53,349 INFO spawned: 'video_looper' with pid 9885
2024-05-03 19:55:58,916 INFO success: video_looper entered RUNNING state, process has stayed up for > than 5 seconds (startsecs)
```

video_looper-stdout constantly repeats the same log entry 5 times per second:

```
==> video_looper-stdout---supervisor-_9joFZ.log <==
[2024-05-03 19:56:56.891073] Playing movie: video.mp4 (video) (endless loop)
[2024-05-03 19:56:57.075756] Playing movie: video.mp4 (video) (endless loop)
[2024-05-03 19:56:57.260718] Playing movie: video.mp4 (video) (endless loop)
[2024-05-03 19:56:57.479736] Playing movie: video.mp4 (video) (endless loop)
[2024-05-03 19:56:57.671416] Playing movie: video.mp4 (video) (endless loop)
```

githubarooski, 2024-05-03:
Oh, and I'm using your distro .zip file directly for the install, not compiling it myself.

tofuSCHNITZEL, 2024-05-03:
hm yes the playing movie entry appears because omxplayer is constantly crashing so pi_video_looper is starting the "next" video...
basically pi_video_looper is just a "fancy" process launcher so if you can run omxplayer manually I cant think of a reason why it would not run over pi_video_looper....  what if you remove the -g from custom_args?

unfurlgit, 2024-06-03:
removing -g doesn't fix the issue. the only tiny outlier I can think of is that when I initially configured the OS, I set the username as an alternative username. I subsequently added the ```pi``` user back to the OS and the script is now running out of /home/pi/ like it expects to. 

```
$ grep extra_args /home/pi/pi_video_looper/assets/video_looper.ini 
extra_args = --no-osd --audio_fifo 0.01 --video_fifo 0.01 --align center --font-size 55
```

```
root@hostname:/var/log/supervisor# tail -f *
==> supervisord.log <==
2024-06-03 15:40:37,807 INFO Included extra file "/etc/supervisor/conf.d/video_looper.conf" during parsing
2024-06-03 15:40:37,875 INFO RPC interface 'supervisor' initialized
2024-06-03 15:40:37,875 CRIT Server 'unix_http_server' running without any HTTP authentication checking
2024-06-03 15:40:37,876 INFO supervisord started with pid 468
2024-06-03 15:40:38,884 INFO spawned: 'video_looper' with pid 478
2024-06-03 15:40:44,299 INFO success: video_looper entered RUNNING state, process has stayed up for > than 5 seconds (startsecs)
2024-06-03 15:44:36,104 INFO waiting for video_looper to stop
2024-06-03 15:44:36,496 INFO stopped: video_looper (exit status 0)
2024-06-03 15:44:36,514 INFO spawned: 'video_looper' with pid 7071
2024-06-03 15:44:42,419 INFO success: video_looper entered RUNNING state, process has stayed up for > than 5 seconds (startsecs)

==> video_looper-stderr---supervisor-RdMFA9.log <==

==> video_looper-stdout---supervisor-jott80.log <==
[2024-06-03 15:44:50.560425] Playing movie: video.mp4 (video) (endless loop)
[2024-06-03 15:44:50.964147] Playing movie: video.mp4 (video) (endless loop)
[2024-06-03 15:44:51.382080] Playing movie: video.mp4 (video) (endless loop)
[2024-06-03 15:44:51.795857] Playing movie: video.mp4 (video) (endless loop)
[2024-06-03 15:44:52.188375] Playing movie: video.mp4 (video) (endless loop)
```

githubarooski, 2024-06-03:
removing -g doesn't fix the issue. the only tiny outlier I can think of is that when I initially configured the OS, I set the username as an alternative username. I subsequently added the ```pi``` user back to the OS and the script is now running out of /home/pi/ like it expects to. 

```
$ grep extra_args /home/pi/pi_video_looper/assets/video_looper.ini 
extra_args = --no-osd --audio_fifo 0.01 --video_fifo 0.01 --align center --font-size 55
```

```
root@hostname:/var/log/supervisor# tail -f *
==> supervisord.log <==
2024-06-03 15:40:37,807 INFO Included extra file "/etc/supervisor/conf.d/video_looper.conf" during parsing
2024-06-03 15:40:37,875 INFO RPC interface 'supervisor' initialized
2024-06-03 15:40:37,875 CRIT Server 'unix_http_server' running without any HTTP authentication checking
2024-06-03 15:40:37,876 INFO supervisord started with pid 468
2024-06-03 15:40:38,884 INFO spawned: 'video_looper' with pid 478
2024-06-03 15:40:44,299 INFO success: video_looper entered RUNNING state, process has stayed up for > than 5 seconds (startsecs)
2024-06-03 15:44:36,104 INFO waiting for video_looper to stop
2024-06-03 15:44:36,496 INFO stopped: video_looper (exit status 0)
2024-06-03 15:44:36,514 INFO spawned: 'video_looper' with pid 7071
2024-06-03 15:44:42,419 INFO success: video_looper entered RUNNING state, process has stayed up for > than 5 seconds (startsecs)

==> video_looper-stderr---supervisor-RdMFA9.log <==

==> video_looper-stdout---supervisor-jott80.log <==
[2024-06-03 15:44:50.560425] Playing movie: video.mp4 (video) (endless loop)
[2024-06-03 15:44:50.964147] Playing movie: video.mp4 (video) (endless loop)
[2024-06-03 15:44:51.382080] Playing movie: video.mp4 (video) (endless loop)
[2024-06-03 15:44:51.795857] Playing movie: video.mp4 (video) (endless loop)
[2024-06-03 15:44:52.188375] Playing movie: video.mp4 (video) (endless loop)
```

githubarooski, 2024-06-03:
tldr, I reformatted the drive with balena etcher which doesn't give you all the tweaking options that the raspberry pi imager provides and now it's working as expected. I think it was the extra settings portion of the raspberry pi imager which dorked everything up.

githubarooski, 2024-06-03:
Ok, sorry for the many notifications, but I was wrong, this is still a problem, even with a fresh install, then upgrade to 1.0.18 with install.sh 

Can you try testing on your end passing a custom resolution via extra_args?

muggsyd, 2024-08-24:
I can replicate this issue, however in a slightly different scenario. I am using a Raspberry pi 3B with a SCART HAT based on VGA666. I am using the image supplied by video-looper.de and upgrading manually to v1.0.19

I can also replicate the black screen when playing via the python script. Using the same arguments as supplied in the .ini I can run it via the shell using omxplayer directly without issue, however when using the video-looper I get a black screen

tofuSCHNITZEL, 2024-08-24:
As which user are you running the omxplayer command? Does it work/not work if run it as root/non root?

muggsyd, 2024-08-24:
When I run the omxplayer command manually, I am only using the "pi" user account via ssh which should be non-root (as I need to elevate via sudo to read the looper logs in /var/log/supervisor

tofuSCHNITZEL, 2024-08-24:
Okay and if you run it as root manually? Does it still work (I would expect not, because video looper runs omx as root)

muggsyd, 2024-08-24:
It actually runs fine as root when playing a single file from /home/pi/video

If I reboot my device, I get no video playback, although I do see the rpi output during the boot process. As it's connected via WiFi I can't capture any htop instances during this time.

If I ssh into the device, and issue a reload command, I see the omxplayer processes spawn and a random video will play successfully. However, when that video finishes playing no new processes are spawned, even though the 

```
$ start
video_looper: ERROR (already started)
$ reload
video_looper: stopped
video_looper: started
$ start
video_looper: ERROR (already started)
$
```

However, during my testing I found some inconsistencies with when omxplayer actually ran. For example, if I issued the "stop" command to prepare for more testing, suddenly the omxplayer process would pop up in htop and play 1 random file and then stop.  There are 6 video files in the video path but it would choose one, and one only randomly to play and then stop. I have also removed all spacing from my video files so they are named 1.avi, 2.avi etc

I have also replicated the "fix" that @githubarooski suggested as below and that works correctly upon reboot. However I don't want to loop one video, I want to loop thru my specified media folder randomly
```
$ cat /etc/supervisor/conf.d/video_looper.conf
# Supervisord configuration to run video looper at boot and
# ensure it runs continuously.
[program:video_looper]
#command=python3 -u -m Adafruit_Video_Looper.video_looper
command=/bin/bash /usr/bin/omxplayer -o local --win "0 0 320 240" -g --loop '/home/pi/video/1.avi'
autostart=true
autorestart=unexpected
startsecs=5
```
What else can I do to try and narrow down what is causing this. I get no errors logs in the stdout and stderr files either.