adafruit/pi_video_looper #221: Add more omxplayer key bindings?

cstoney, 2024-05-13:
I would love to have more omxplayer controls available. Specifically, the ability to seek within videos:

```
left arrow  seek -30 seconds
right arrow seek +30 seconds
down arrow  seek -600 seconds
up arrow    seek +600 seconds
<           rewind
>           fast forward
z           show info
```

🙏

lukethemaker, 2024-09-04:
Chiming in to say I would also like the playback speed control keys available (1 faster and 2 slower)

falue, 2024-09-25:
I chime in too and i'd love to have spacebar for play/pause!

tofuSCHNITZEL, 2024-09-25:
> I chime in too and i'd love to have spacebar for play/pause!

This is already working.

falue, 2024-09-25:
> > I chime in too and i'd love to have spacebar for play/pause!
> 
> This is already working. 

Thanks for the fastest reply! Where is this documented? I only see esc, k, p, and s as possible keyboard inputs in the docs, and spacebar does nothing on my pi.
I followed the tutorial by videolooper.de if that matters?

falue, 2024-09-25:
> > I chime in too and i'd love to have spacebar for play/pause!
> 
> This is already working. 

I see - i am using the precompiled image from videolooper.de, updated the codebase manually and assumed it would work. It does not so i will be following the official guideline for installing. Thanks for the reply!

pelicanmedia, 2024-09-26:
Is there just a stop key bind not the stop/play option? (_"s" - Stop/Start - stops or starts playback of current file_)
I'd like to stop the video I'm playing and wait for the play command again.
eg I have `gpio_pin_map = "13": "1.mp4", "15": "K_s"`
If pin 15 is triggered the video stops correctly, but if it's triggered again the video plays again, I only want it to start again if pin 13 is triggered.