adafruit/pi_video_looper #222: one file in the video folder

ozgur-as, 2024-05-20:
hi there,

(using 2.8 image from the website with the looper version 1.0.17)
i'm trying to setup a basic one-shot button player with gpio and having a problem.

unless there are 2 files in the media folder, the video loops. 

the changed settings are below:
file_reader = directory
osd=false
one_shot_playback = true
bg_image = /home/pi/splash.png
gpio_pin_map = "5" : "video.mp4"
sound = hdmi

if there are 2 files named video.mp4 and another file in the media directory, it works fine (play once when button is pressed), if there is only video.mp4 in the directory, the file loops.

another "not so important" error is that the video plays once when the app starts even though there are no triggers from the gpio.

thanks in advance

tofuSCHNITZEL, 2024-05-20:
Hi,
thanks for pointing that out!
as soon as there is only one file omxplayer is called with the loop argument and the python script will not to any checks again since the omxplayer process will never exit.
I will have a look if this is an easy fix.

since this script started as a video looper, playing a video on startup is just by design. 
I will see if I can add a "dont play on startup" setting.

cheers

tofuSCHNITZEL, 2024-05-20:
The looping issue is fixed in "next" https://github.com/adafruit/pi_video_looper/tree/next
and I have also added a "play on startup" setting.
not sure if I will have time today to test the changes - so please test it if you have the time.

tofuSCHNITZEL, 2024-05-20:
all tested and published in version v1.0.18

https://github.com/adafruit/pi_video_looper/releases/tag/v1.0.18

closed 2024-05-20 by tofuSCHNITZEL