adafruit/pi_video_looper #228: Copy Files

philspitlerSF, 2024-08-06:
Hi, I am using Videolooper 2.8 and it works great reading my files from the USB drive.

When I enable copy mode, it starts to copy the file but once it get's to about 50% of the copy, it reboots and gets stuck in a cycle.

The mp4 file is 765mb.

tofuSCHNITZEL, 2024-08-06:
Hi,
sounds to me like a corrupt file / sd card / usb stick. 
have you tried with a different file / sd card (fresh flash) / usb stick?

christiansievers, 2024-08-06:
Yup. Or there isn’t enough space on the SD card. Have you expanded it? As usual with raspi-config.

tofuSCHNITZEL, 2024-08-06:
> Yup. Or there isn’t enough space on the SD card. Have you expanded it? As usual with raspi-config.

ah yes of course - @philspitlerSF what is the output of "lsblk" and "df -h"

philspitlerSF, 2024-08-06:
The file / USB are not corrupt but it could be a file size  issue. I thought I read that the file system was already expanded in the image file I burned.

I'll test it using the comments suggested here and report back.

philspitlerSF, 2024-08-06:
It was a file size issue, I misread the documentation and thought it said the file system had already been expanded. It works great now. Thanks

tofuSCHNITZEL, 2024-08-06:
Nice!

closed 2024-08-06 by tofuSCHNITZEL